StephanJoubert/home_assistant_solarman #1: Home Assistant Energy Config

tuanha2000vn, 2022-01-06:
Thank for the AWESOME custom component... been looking for a while and that exactly what I need.

Do you use the Energy configuration in Home Assistant? What is the entity for Grid consumption, Return to grid, Battery system, Solar production?

[image]

The Solarman component add 50 entities for me
```
type: entities
entities:
  - entity: sensor.deye_ac_temperature
  - entity: sensor.deye_alert
  - entity: sensor.deye_battery_current
  - entity: sensor.deye_battery_power
  - entity: sensor.deye_battery_soc
  - entity: sensor.deye_battery_status
  - entity: sensor.deye_battery_temperature
  - entity: sensor.deye_battery_voltage
  - entity: sensor.deye_communication_board_version_no
  - entity: sensor.deye_control_board_version_no
  - entity: sensor.deye_current_l1
  - entity: sensor.deye_current_l2
  - entity: sensor.deye_daily_energy_bought
  - entity: sensor.deye_daily_energy_sold
  - entity: sensor.deye_daily_load_consumption
  - entity: sensor.deye_daily_production
  - entity: sensor.deye_dc_temperature
  - entity: sensor.deye_external_ct_l1_power
  - entity: sensor.deye_external_ct_l2_power
  - entity: sensor.deye_gen_connected_status
  - entity: sensor.deye_gen_power
  - entity: sensor.deye_grid_connected_status
  - entity: sensor.deye_grid_status
  - entity: sensor.deye_grid_voltage_l1
  - entity: sensor.deye_grid_voltage_l2
  - entity: sensor.deye_internal_ct_l1_power
  - entity: sensor.deye_internal_ct_l2_power
  - entity: sensor.deye_inverter_id
  - entity: sensor.deye_inverter_l1_power
  - entity: sensor.deye_inverter_l2_power
  - entity: sensor.deye_load_l1_power
  - entity: sensor.deye_load_l2_power
  - entity: sensor.deye_load_voltage
  - entity: sensor.deye_micro_inverter_power
  - entity: sensor.deye_pv1_current
  - entity: sensor.deye_pv1_power
  - entity: sensor.deye_pv1_voltage
  - entity: sensor.deye_pv2_current
  - entity: sensor.deye_pv2_power
  - entity: sensor.deye_pv2_voltage
  - entity: sensor.deye_running_status
  - entity: sensor.deye_smartload_enable_status
  - entity: sensor.deye_time_of_use
  - entity: sensor.deye_total_energy_bought
  - entity: sensor.deye_total_energy_sold
  - entity: sensor.deye_total_grid_power
  - entity: sensor.deye_total_grid_production
  - entity: sensor.deye_total_load_consumption
  - entity: sensor.deye_total_load_power
  - entity: sensor.deye_total_power
  - entity: sensor.deye_total_production
  - entity: sensor.deye_work_mode
title: SolarMAN
```

StephanJoubert, 2022-01-07:
Thank you for the kind words. 

I don't have much experience with the Energy portion of HA, but the way you have configured your system looks correct to me.

tuanha2000vn, 2022-01-10:
[image]

Thank again, the custom component provide all detail sensor, after a few tweak, I've managed to make this awesome energy monitor <3

StephanJoubert, 2022-01-11:
Very impressive. I have been trying with the energy monitor with a little success. Would you mind sharing some pointers how you got the very nice looking dashboard.

tuanha2000vn, 2022-01-13:
[image]

Most of the work have been done by your custom component.
Grid consumption: DEYE Daily Energy Bought
Return to grid: DEYE Daily Energy Sold
Solar production: DEYE Daily Production

I've only add 2 custom sensor for battery in and out.

Battery data provide in sensor.deye_battery_power, negative number mean charging, positive number mean discharging. So I've create 2 sensor template for number of W charging/discharing like this:
```
- platform: template
    batt_in:
      friendly_name: "Batt In W"
      value_template: >
        {%- if states.sensor.deye_battery_power.state | int <= 0 -%}
        {{ 0 - states.sensor.deye_battery_power.state | int }}
        {%- else -%}
        0{%- endif %}
      unit_of_measurement: W
      device_class: power
      icon_template: mdi:battery-plus
    batt_out:
      friendly_name: "Batt Out W"
      value_template: >
        {%- if states.sensor.deye_battery_power.state | int > 0 -%}
        {{ states.sensor.deye_battery_power.state }}
        {%- else -%}
        0{%- endif %}
      unit_of_measurement: W
      device_class: power
      icon_template: mdi:battery-minus
```

Use platform integration to convert W to kWh
```
- platform: integration
  source: sensor.batt_in
  name: batt_in_integration
  unit_prefix: k
  round: 3

- platform: integration
  source: sensor.batt_out
  name: batt_out_integration
  unit_prefix: k
  round: 3
```
Then use unility_meter to reset it to zero every day
```
utility_meter:
  batt_in_integration_daily:
    source: sensor.batt_in_integration
    cycle: daily
  batt_out_integration_daily:
    source: sensor.batt_out_integration
    cycle: daily
```
Finally Configuration > Energy > Add Battery system
Energy going into the batterry (kWh): sensor.batt_in_integration_daily
Energy coming out of the batterry (kWh): sensor.batt_out_integration_daily

timon65za, 2022-02-09:
> [image]
> 
> Most of the work have been done by your custom component. Grid consumption: DEYE Daily Energy Bought Return to grid: DEYE Daily Energy Sold Solar production: DEYE Daily Production
> 
> I've only add 2 custom sensor for battery in and out.
> 
> Battery data provide in sensor.deye_battery_power, negative number mean charging, positive number mean discharging. So I've create 2 sensor template for number of W charging/discharing like this:
> 
> ```
> - platform: template
>     batt_in:
>       friendly_name: "Batt In W"
>       value_template: >
>         {%- if states.sensor.deye_battery_power.state | int <= 0 -%}
>         {{ 0 - states.sensor.deye_battery_power.state | int }}
>         {%- else -%}
>         0{%- endif %}
>       unit_of_measurement: W
>       device_class: power
>       icon_template: mdi:battery-plus
>     batt_out:
>       friendly_name: "Batt Out W"
>       value_template: >
>         {%- if states.sensor.deye_battery_power.state | int > 0 -%}
>         {{ states.sensor.deye_battery_power.state }}
>         {%- else -%}
>         0{%- endif %}
>       unit_of_measurement: W
>       device_class: power
>       icon_template: mdi:battery-minus
> ```
> 
> Use platform integration to convert W to kWh
> 
> ```
> - platform: integration
>   source: sensor.batt_in
>   name: batt_in_integration
>   unit_prefix: k
>   round: 3
> 
> - platform: integration
>   source: sensor.batt_out
>   name: batt_out_integration
>   unit_prefix: k
>   round: 3
> ```
> 
> Then use unility_meter to reset it to zero every day
> 
> ```
> utility_meter:
>   batt_in_integration_daily:
>     source: sensor.batt_in_integration
>     cycle: daily
>   batt_out_integration_daily:
>     source: sensor.batt_out_integration
>     cycle: daily
> ```
> 
> Finally Configuration > Energy > Add Battery system Energy going into the batterry (kWh): sensor.batt_in_integration_daily Energy coming out of the batterry (kWh): sensor.batt_out_integration_daily
> 
> [image] I've also add Configuration > Intergration > Add Integration Forecast.Solar for a nice prediction graph which display in the Energy section like this

Thanks for the details but can you please help in where you add the scripting.
I have tried putting it in configuration.yaml but I am getting a lot of errors.
Sorry for the dum questions but I am new to HA and it is a new installations.
I did had the energy working until HA upgraded to 2022.2.3 now it is complaining about "Grid neutrality could not be calculated"
Hopefuuly with youre work around i can get it fixed

StephanJoubert, 2022-02-09:
@timon65za The scripts (templates) goes in configuration.yaml, but I also had problems. It seems that HA changed some of the specifications with the updates and one need to make provision for that. I think you can start with the scripts as provided by @tuanha2000vn , and then google the help for "templates" to see if you can fix the errors. 

I plan to add scripts and blueprints in the repo that people can just use, but need to test it first to see if it works. One blueprint will enable my swimming pool pump only while I am not working off the battery. Almost like a non-critical load, but bit more intelligent.

kolakidd, 2022-02-22:
I think I solved the above question on placement and formatting:

I made to the code was the addition of sensors: in the platform template. that got that part working. You just have to watch the formatting when cutting and pasting.

Placement is as follows:

On it's own, as in at the top of configuration.yaml or somewhere where it's not inside sensors, binary_sensors etc:

```
utility_meter:
  batt_in_integration_daily:
    source: sensor.batt_in_integration
    cycle: daily
  batt_out_integration_daily:
    source: sensor.batt_out_integration
  cycle: daily
```

 
Then, under sensor:

```
  - platform: template
    sensors:
      batt_in:
        friendly_name: "Batt In W"
        value_template: >
          {%- if states.sensor.deye_battery_power.state | int <= 0 -%}
          {{ 0 - states.sensor.deye_battery_power.state | int }}
          {%- else -%}
          0{%- endif %}
        unit_of_measurement: W
        device_class: power
        icon_template: mdi:battery-plus
      batt_out:
        friendly_name: "Batt Out W"
        value_template: >
          {%- if states.sensor.deye_battery_power.state | int > 0 -%}
          {{ states.sensor.deye_battery_power.state }}
          {%- else -%}
          0{%- endif %}
        unit_of_measurement: W
        device_class: power
        icon_template: mdi:battery-minus
  - platform: integration
    source: sensor.batt_in
    name: batt_in_integration
    unit_prefix: k
    round: 3
  - platform: integration
    source: sensor.batt_out
    name: batt_out_integration
    unit_prefix: k
    round: 3
```

@StephanJoubert thank you for this amazing/lekka work on the integration.

havoc2021, 2022-02-24:
Hi @StephanJoubert
Thank you for this AWESOME work, I finally got my Deye inverter in HA and all data displaying correctly been looking for weeks for something like this.

@tuanha2000vn  thank you for the only missing piece I needed, one problem and I hope you can help me with it. I could get the battery display in Energy dashboard, but seems like nothing is flowing to the battery but it's currently charging.
[image]

kolakidd, 2022-02-24:
It took a while for the data to start logging, I think the energy dashboard takes around two hours to populate when you first set it up. The way I checked if it was working was to set up an entity card with battery in and battery out, then click on either of those and you will see a small graph that will show if they are logging power

Entities:
[image]

What you should see when you click on the entity:
[image]

havoc2021, 2022-02-24:
Hi @kolakidd 

You correct after like 30min or so it started showing up :-D

[image]

THANK YOU ALL!!!

StephanJoubert, 2022-02-27:
@kolakidd . @timon65za , @tuanha2000vn Just to let you know that I dug around the (very scarce) documentation and found the modbus registers for the battery charge and discharge from the inverter. It is therefore not needed to do the template calculation anymore. My expectation is that this would be a much more  accurate method if there are large battery charge/discharge cycles. It forms part of release 1.3.0 so you can just update using HACS.

kolakidd, 2022-02-27:
@StephanJoubert awesome! Updated now, will give it a try.

havoc2021, 2022-02-27:
@StephanJoubert hope you dont mind, I have given it a go as well :-D

havoc2021, 2022-02-28:
Hi @StephanJoubert 
Thank you for the charge and discharge sensor, I take it this replace the IN/OUT template.
But I see this is TOTAL charge and discharge, anyway you can make it daily?

StephanJoubert, 2022-02-28:
@havoc2021 , yes this takes the place of the template. Unfortunately this is the only values provided by the inverter, but the  state_class "total_increasing" tells the home assistant dashboard to mark the value at midnight and calculate the daily values. I use on my system, and the dashboard  and graphs are correct. I have updated the documention with the new entities:
See: https://github.com/StephanJoubert/home_assistant_solarman/blob/main/energy.md

If one needs a daily value for another reason than the dashboard, it could be done with the "utility_meter" component from what I can see - have not done that myself.

kolakidd, 2022-02-28:
Just as a note, I updated and then tracked the four values (Modbus source VS Template) in Graphana. The template values are very good, but not 100% accurate. For instance there is a slight overlap between charge and discharge, which shouldn't be possible, because a battery can't simultaneously charge and discharge. This is not to say that they won't give you a good idea.

I've updated to the Modbus values on the dashboard this morning.

closed 2022-03-23 by StephanJoubert