StephanJoubert/home_assistant_solarman #3: Serial Number with Letters

ebarretopr, 2022-01-21:
My inverter Sol-ark (US) dongle serial number has letters in the beginning and its not accepted as a entry in YAML

StephanJoubert, 2022-01-22:
It is possible that the logger is incompatible with the protocol. What you can try is to go onto the web portal (using a browser to navigate to the IP, and use "admin" "admin" as username and password. On the status screen expand "Device information" and note the Device serial number. I included a screen capture:
[image]
.

ebarretopr, 2022-01-23:
My apologies mine is completely different from yours

[image: image.png]

closed 2022-01-29 by StephanJoubert