StephanJoubert/home_assistant_solarman #4: Not exposed in HA integrations

dswarne, 2022-01-29:
Hi,

Firstly thanks for the efforts, much appreciated.
Could you maybe assist me? I have successfully downloaded the integration in HACS and i can see it in the relevant dashboards and file system, however the plugin is not exposed in HA integrations when I search.

Am I missing something obvious or doing something wrong?

Kind Regards

StephanJoubert, 2022-01-29:
Hi,

I think you are almost there. You can try the following:
1. Click on "Developer Tools" on the left, and under "States" view the entities.
2. Under "Filter Entities" type "inverter". If the system works correctly you should see a number of entities ex.sensor.deye_inverter_id (in my case I called the inverter deye, but it may differ depending on  your configuration)

The screen capture in the readme was created by adding a card "by entity", and under "Search Entities" I just filtered by typing the name "deye" and selected multiple entities. The name "deye" was pulled from my configuration and may be different depending on your configuration.

Hope you have success.

dswarne, 2022-01-29:
Thanks for the feedback, I have however only downloaded the integration but can start it or even find the config to edit the ip address, I'm relatively new to custom plugins and I'm sure I've missed a step.

All i did was find the integration in HACS then click the download. cant figure out the next step.

StephanJoubert, 2022-01-29:
You need only add a few steps:
1. You need to configure the router to give the inverter a fixed IP. This is important since the component will try to find the inverter at a specific IP.
2. Then get the serial number of the inverter-collector. The easiest way to do that is to open a browser and browse to the IP you specified above. The login with username "admin" and password "admin". On the web UI, under "Status" expand "Device information" and note the "Device serial number"
3. Then you need to add a few lines in the configuration.yaml file. There is a configurator that makes this relatively easy.

In that file, you just need to add an entry under "sensors" as follows:
sensors:
  - platform: solarman
    name: SUNSYNK <- you can change this to the brand of the inverter
    inverter_host: 192.168.0.201 <-- you need to change this to the IP you specified in step 1
    inverter_port: 8899
    inverter_serial: 1720747149 <-- change this to the serial number you got in step 2
    scan_interval: 30
4. Then restart Home Assistant to accept the changes.

dswarne, 2022-01-29:
Thanks Bud,

All visible now, I made a noob error by "sensors" and not "sensor in the config

xb385, 2022-01-29:
To add to this, i got all the entities showing but everything is set to "unknown" values
as far as i can tell everything is configured correctly

Toma82, 2022-01-30:
I have the same problem in HA no entities are unknown 
[image]

And in integration after I have the following error

[image]

StephanJoubert, 2022-01-30:
The message trace refers to a different piece of code. A quick google seems to point to here: https://github.com/MichaluxPL/Sofar_LSW3

It may be that the two components influence each other so I would recommend disabling the Sofar_LSW3 code  just to see if it makes a difference.

The good news is that both the LSW3 integration and this component uses the same protocol, so if one of them works, the other one should also work.

Toma82, 2022-01-30:
Well, unfortunately, despite the removal of that integration, I still have unknown entities
[image]

xb385, 2022-01-30:
i still dont get a read out, where exactly can i see errors so i can try to bugfix?

StephanJoubert, 2022-01-31:
You can look at the logs (Configuration | Settings | Logs) and click on "Load full home assistant logs" and see if you get any traces.

I have updated the readme.md file to help with installation. You can just go though that process as well to confirm that the configuration is correct. If the collector you have is incompatible, you can try the "Omnik Solar integration" that is newly added to HACS to see if that works.

ianfretwell, 2022-01-31:
The example code incorrectly states that the config needs to go under 'sensors:'  as pointed out above, this should actually be 'sensor:'

Regardless of that I seem to have the same issue as @xb385 - the integration has brought all the sensors in but they are all marked as 'unknown' - although very, very occasionally I see two of them change to 'LOOKUP'

This integration works just fine though (I have it disabled whilst I am trying to get yours working) [https://github.com/robbinjanssen/home-assistant-omnik-inverter](url)
This also appears to be using v5 of the connection so I really can't see any reason why yours won't work. Any more ideas ?

StephanJoubert, 2022-01-31:
Thanks for pointing me to the typo in the documentation - I have fixed and pushed that.

I had a look at the other integration, and from what I can gather it uses a different method to obtain the data. Rather than using the modbus interface, it queries it directly from the web JSON API. So it is stil possible that the version is incompatible.

Apart from the version of the protocol, we also have the mapping of parameter ids to actual parameters. Ex 00BD being Battery Status. I have verified that against the solarman app, but in case Omik has a different app it may use a  different mapping and be incompatible?

donovanvdberg, 2022-01-31:
Hi All,  Anyone got this working with Firmware version LSW3_15_FFFF_1.0.65 ?
IP has been updated and set static and SN updated in config file. All Sensors are stating Unknown

ianfretwell, 2022-01-31:
@StephanJoubert  I missed your point about Solarman app - yes, I can use this to see my details just fine - no issues there.

It looks like some of the figures are updating - but it's now at a consistent ten minute interval (not 30 seconds, which is what is in the config) : -

[image]

ianfretwell, 2022-01-31:
@StephanJoubert OK, many thanks anyway - it was worth a go!

closed 2022-02-06 by StephanJoubert