StephanJoubert/home_assistant_solarman #5: No Data or Unique ID

jimmy-mc, 2022-01-30:
Firstly - this is my potential unicorn! Been looking for so long to try and get my SofarSolar ME3000SP into Home Assistant..

I can split this if it’s two issues, but I’m not getting any figures returned. The IP/serial is correct, and I also get the following on all entities - is it related to my issue?

“This entity ('sensor.solarinverter_daily_production') does not have a unique ID, therefore its settings cannot be managed from the UI. See the documentation for more detail.”

Could you give me any pointers as to where I’d look for errors? Nothing in the main HA log, but any others I should check?

Many thanks, and appreciate your work..

StephanJoubert, 2022-01-30:
I just had a look to see if I can replicate the issue, and if I select the entity, then click on the gear icon, I get the same message. However, the entity is updated automatically and you do not need to manage the entity settings.

If you select the entity on a card, and it displays the actual value (instead of "unknown"), the component is working as expected. If however it stays on "unknown" it may be due to the following:

1. Check if the inverter has a solarman wifi stick.  I do not have a Sofar inverter, but from info I received it does seem like there is a high probabilty that the collector will be compatible.
https://www.sofarsolar.com/product-detail/499/Stick-Logger
2. Verify the settings. For that go the web portal of the stick on the local network (not the solarman web portal). If it asks for login, it is proably "admin" and "admin". Then expand "Device Information" and note the "Device serial number". The IP you used and the serial number displayed in the portal is supposed to be used in the configuration settings.
3. If it works and you get the information you need, just reserve the IP for the stick on the router so that it will not change.

[image]

jimmy-mc, 2022-01-30:
Thanks for the quick reply. That side is all good - I’ve had it working with Solarman for a year now. One difference seems to be that you’re on a slightly higher firmware of the logger (1.29) whereas I’m on 1.15. Wonder if that could be an issue. I’ll see if I can track down some firmware for it..

Other than that, it’s pretty much the same, hence wondering if there are any logs or trace info somewhere.

Thanks though

jimmy-mc, 2022-01-30:
Update - my serial number starts 40, which I think might use the earlier version of the Solarman protocol, so think I’ll need to try that now archived project you linked to. Might get a more recent logger and hook that in, in the medium term

StephanJoubert, 2022-01-30:
I see someone else took over that project, and it is now added to HACS. Good luck!

https://github.com/hultenvp/home_assistant_omnik_solar

closed 2022-01-30 by StephanJoubert