StephanJoubert/home_assistant_solarman #6: Enhancement: Include Debug Messages and Configuration Options for the Registers

fabiobrolo, 2022-01-31:
Hi, your integration is the best that I´ve found in the internet to get the local measurements. Thanks for the effort to develop.
I have a DEYE 8K inverter using a logger with firmware LSW3_15_FFFF_1.0.72 and most of the measurements are not being populated with correct values.
Maybe the registers for my logger is different from yours.
Is it possible to include debug messages to show the raw message captured from the logger and also include a config to map the registers?
How you figured out the correct registers for your logger? Maybe I can try the same approach and can contribute with this enhancement.
Thanks a lot.

StephanJoubert, 2022-02-01:
Thanks for the kind words. 

It would be great if you want to contribute. To experiment with the parameters it is relatively easy. The parameter mapping  (ID to name and type) is defined in  Parameters.yaml. "custom_components/solarman/parameters.yaml). Then the values requested is at the bottom of the file solarman.py as "self.send_request (params, 0x0003, 0x000E)" 

I am looking at ways to log the info in a sensible way and will include that with the next update.

As far as the parameters go, I reverse engineered the solarman pro mobile app and got the definitions from it. See attached zipfile. I used the file DYRealTime.txt as the source and that  maps correctly for the 5kW unit. Have a look at "RealTimeData.txt" which includes simlilar data, but with different register allocations. Let me know if those register allocations yield better results.

[assets.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/7979372/assets.zip)

andrempo, 2022-02-03:
I have a similar issue with the SUN-3K-G inverter. Just some values are correct.
[image]
I've tried to check "RealTimeData.txt" but the values are very different. (the ones that are working are not the same)
Do you have any idea where can I find the correct values?

StephanJoubert, 2022-02-06:
@fabiobrolo . Although I don't have the specific field definitions for the 8K inverter, I added the ability to tweak the parameter definition by only editing the parameters.yaml file. All the hardcoded requests are now removed. If you feel up to it to experiment have a look at the latest release, and see https://github.com/StephanJoubert/home_assistant_solarman/blob/main/customization.md

If you have success please let me know.

StephanJoubert, 2022-02-15:
@andrempo , It seems you have the 3-phase inverter. There is  now a specific issue for this so please have a look here -> https://github.com/StephanJoubert/home_assistant_solarman/issues/17

andrempo, 2022-02-24:
> @andrempo , It seems you have the 3-phase inverter. There is now a specific issue for this so please have a look here -> #17

Great! Now it is working for me.
Thanks!

fabiobrolo, 2022-03-04:
Hi @StephanJoubert 
The deye string configuration file is working great for my 8K Deye inventer.
The only missing data is the PV1 Power and PV2 Power.
Thanks a lot for the enhancement.

closed 2022-03-04 by fabiobrolo