StephanJoubert/home_assistant_solarman #7: Cheeky Question

dswarne, 2022-02-01:
Hey Stephan, this integration works beautifully, hence the following cheeky question:

Would you be able to re create the package for a "Homebridge" install?

Sorry for asking but i had to ask. As it would be very popular.

StephanJoubert, 2022-02-01:
I had a quick look and it looks cool, but I am not using the Apple ecosystem so won't be able to test. However, I can write a typescript object to be used in NodeJS and someone with Apple/Homebridge experience can then do the connection.

Do you know if homebridge can be integrated with a Nest HUB?

dswarne, 2022-02-01:
Hi,

I can do all the testing and supporting from a HomeKit/Homebridge side.

Re- the nest hub integration, it is possible with Homebridge: below Reddit
can assist.

https://www.reddit.com/r/homebridge/comments/e51p8o/received_a_google_nest_hubwhat_can_i_use_it_for/

Kind Regards

dswarne, 2022-02-01:
Just some more info:

https://assistant.google.com/services/a/uid/000000b558f0d5d1?hl=en

Regards

StephanJoubert, 2022-02-01:
Thanks. Will let you know as soon as I start the project.

dswarne, 2022-02-01:
Fantastic news