StephanJoubert/home_assistant_solarman #10: Sensors showing up as graphs instead of state changes

hyarionh, 2022-02-02:
I wanted to make a pull request to fix this but my Python is fairly rudimentary and I'm sure there's better ways to fix it than my "hack" job so perhaps a discussion is best.

Currently all sensors show up in home assistant as a "number" style sensor.  i.e. if you click on it, it shows a graph.  This is fine for the majority of sensors, but for any with a state, e.g. "On-Grid" vs "Off-Grid", it results in it not recording the history because it's expecting a number, not text.

After a lot of experimenting, it seems that it's related to the `unit_of_measure` property.  It does not matter what value this property has, if it exists then Home Assistant assumes it has a number for values.

My crude way of fixing this so far has involved making a copy of the `SunsynkSensor` class giving it a name like `SunsynkSensorText` and it leaves out the `unit_of_measurement` as well as `extra_state_attributes`.  I then added a `isstr` to the parameters.yaml to split the sensors up.

While this works, it doesn't feel very elegant.  But perhaps that's ok?

@StephanJoubert let me know if you'd like me to submit a pull request for this as-is or recommend a better way for me to do it, or if you'd like to tackle it yourself.  Assuming it's something that bugs you too and would like to fix it :)

StephanJoubert, 2022-02-02:
I have not yet looked at the pull request, but the proposal is not bad at all.

hyarionh, 2022-02-03:
Ok great, I'll submit a pull request with the changes.

closed 2022-02-03 by hyarionh