StephanJoubert/home_assistant_solarman #11: Energy dashboard showing negative at start of new day

hyarionh, 2022-02-02:
I don't know if this only affects Sunsynk or if it also affects DEYE inverters (I have a Sunsynk), but when adding the sensors to the energy graph it works great only until the next day because the inverter then resets "daily production" to zero at midnight. This causes the Home Assistant dashboard to start the day off in a negative.  This has affected me on both solar production and the battery charge/discharge.

To fix this, the `last_reset` value needs to be updated to a new timestamp when the inverter resets the value around midnight.  This is the code I'm using currently:

```
def update(self):
        self.inverter.update()

        val = self.inverter.get_current_val()
        if val is not None:
            if self._field_name in val:           
                self.p_state = val[self._field_name]
                if self._resettable == 1 and self.p_state == 0 and datetime.now().hour == 0:  
                    self._last_reset = datetime.now()
```

I'm doing a very crude check of if the sensor is a resettable one (from my parameters.yaml) and it's value retrieved is zero, and the hour is currently midnight, then update the last_reset to current time.

This is not a perfect solution, if someone can suggest a more robust way of resetting this I'm happy to create a pull request

StephanJoubert, 2022-02-02:
I initially had the same problem and realized I misunderstood the way the dashboard works. Instead of using the daily values, I had to use the total values. After changing that the dashboard worked perfectly.

I suppose the error arise when the total was let's say 10 at 12:00, and then 0 at 0:10, it will then assume it had a huge negative input at 12:00.

The code proposal would work for the daily totals (and may be a good enhancement) and I will keep in mind. Let me know if the totals work for you.

dannyplace, 2022-02-02:
Hi there, 

I had this problem to when using the daily value. Indeed you can use the total value because it is cumulative. But to solve the problem with daily value you could create a utility meter in your config.yaml or when split in other .yaml

You can use the utility meter in your energy dashboard. The cycle switch is the key to reset counter to 0 everyday at 00:00.

```
utility_meter:
  energy:
    name: Solarman Dagelijkse Productie
    #put your sensor here
    source: sensor.template_solarman_invert_pcc_ac1
    cycle: daily
```

hyarionh, 2022-02-03:
Thank you for the comments.  Those are both useful workarounds and I thought they would be sufficient but it seems my battery charge/discharge is not supported (Sunsynk uses a daily cumulative charge and discharge).  The utility_meter solution unfortunately doesn't support the battery side of things.

I have an idea on how to improve the code I posted previously, I'll work on it and provide some feedback

StephanJoubert, 2022-02-03:
@hyarionh Have a look at this thread for a method to handle the battery charge and discharge. tuanha2000vn actually got the dashboard working before I did. If you can confirm that, we can add to energy.md file.
https://github.com/StephanJoubert/home_assistant_solarman/issues/1

hyarionh, 2022-02-03:
@StephanJoubert Ah I see that thread he got it working for the battery.  I'll experiment with it.  The official documentation indicated the `utility_meter` section was only for the grid side.