StephanJoubert/home_assistant_solarman #12: Sunsynk vs DEYE sensors

hyarionh, 2022-02-02:
It appears there's a few slight variations in the values returned from a SunSynk inverter vs a Deye.

e.g. I had to disable "Battery Status" and "Grid Status" as these did not work for my 5kW Sunsynk inverter.

Would it be worth adding a flag for parameters and the config to support specifying which Inverter configuration to use?

e.g. something like:
```
sensor:
  - platform: solarman
    name: SunSynk
    inverter_type: sunsynk    
```

StephanJoubert, 2022-02-02:
I has similar thoughts to handle different inverters. My idea was to have a setting like you mention, and having that map to a specific yaml definition file. In that case someone with a slightly different inverter can modify that specific yaml file .

StephanJoubert, 2022-02-15:
@hyarionh . It took some time, but I have added the abilty to specify the inverter-lookup. (under the config setting "lookup_file").

I have not made a release, so if you want to try, just pull the latest from the repo.

hyarionh, 2022-02-16:
@StephanJoubert, the commits look great!  I will pull it and test it out on my system when I get a chance, hopefully in the next few days.   Thank you for your work on this plugin, it's really appreciated!

closed 2022-02-16 by hyarionh

rmupfumira, 2023-01-20:
> It appears there's a few slight variations in the values returned from a SunSynk inverter vs a Deye.
>
> e.g. I had to disable "Battery Status" and "Grid Status" as these did not work for my 5kW [...]

Please help with how to configure sunsynk for this addon, I just get 'unavailable' on all my sensors