StephanJoubert/home_assistant_solarman #14: Communication Failure detection

timon65za, 2022-02-03:
is there a way that I can detected when there is no communication to the inverter/dongle.
I have done a test by unplugging the dongle, I saw it hold the old values.
Sorry my knowledge is limited to programming, but maybe a date time stamp from the last successful read?
I am pulling the SOC in Node Red which I switch geysers/ swimming pool pumps on and off.
I just need a way to detect that the value is true and healthy.

Thanks for the integrations I had it serial Modbus before but that is wires etc. the wifi is a much neater solution!

StephanJoubert, 2022-02-04:
I am planning to add "meta" entities that would not contain the inverter status, but rather status of the component like communication errors and time of last communication.

timon65za, 2022-02-04:
Thank you that will help allot

StephanJoubert, 2022-02-06:
I have added the entities and updated the documentation as well.

closed 2022-02-06 by StephanJoubert