StephanJoubert/home_assistant_solarman #15: Enable setup via UI

YodaDaCoda, 2022-02-04:
Hi mate, I was working on basically the same thing a couple weeks ago. Looks like your component is a bit better (my first time attempting to make a component for Home Assistant). I did get config happening via UI though - feel free to copy if it'll help. Check out the `config_flow.py` file.

https://github.com/YodaDaCoda/hass-solarman-modbus/

StephanJoubert, 2022-02-04:
Thank you for the info I will certainly use it. I read about the config flow recently and this will help a lot.

jmccrohan, 2022-02-04:
Sorry to intrude on this issue, but I think the chat that @YodaDaCoda and myself had might be useful here:
https://github.com/jmccrohan/pysolarmanv5/issues/1

Consolidating our collective efforts on a single HA integration for all Solarman V5 inverters makes a lot of sense.

closed 2022-03-23 by StephanJoubert