StephanJoubert/home_assistant_solarman #16: Not an issue

felipebraz, 2022-02-06:
This is not an issue, in fact I just crying for help...
I have two Deye Micro Inverters with serial 21xxxx and onboard wifi. Which protocol is it (4, 5,...)? I really want to integrate them with home assistant but is really hard to find something useful... The docs about micro/inverters online is just nonexistent.

StephanJoubert, 2022-02-07:
Hi @felipebraz , I know what you are talking about, the information is scattered over the forums, and I eventually realized that much of it came from the mobile app "solarman pro".

https://play.google.com/store/apps/details?id=com.igen.solarmanpro

The fist test I would do is to see if the device listens on port 4889 (UDP) and  8899. To check, download "Docklight Scripting (https://docklight.de/downloads/), and send  the following to the device at port 48899 on UDP (not TCP)
"WIFIKIT-214028-READ". if that is successful, you should see the device responding with it's IP and serial number.

The second test: Download the solarman pro app to a phone and see if you can access the inverter in any way in "local mode". If so, chances are good that it can be controlled with this component. If not one has to look further with more advanced tools like wireshark.

Let me know if you have any success and we can take it from there.

felipebraz, 2022-02-07:
> Hi @felipebraz , I know what you are talking about, the information is scattered over the forums, and I eventually realized that much of it came from the mobile app "solarman pro". [...]

Hi! About the test:

1- I did it in Packet Sender (it´s the same thing, but free) and worked just fine. One another thing, where is the commands list for there devices? I couldn't find any docs about it.
[image]

2- I couldn't do it just yet because I have Solarman Home, the project in Pro is with my engineer and when I´ve tried to register in Pro it keeps asking me for company certificates and other stuff that I don't have. Is there another way or test I could do?

Thank you for everything.

StephanJoubert, 2022-02-09:
The tests you did look very promising. What I would do is to create a small python app to just issue raw requests with the solarman protocol, and then see if I get a response back. Here is a python module that operates on a low-level hat you can use to create a single request and see that you receive the response.
https://github.com/jmccrohan/pysolarmanv5

StephanJoubert, 2022-02-09:
Re the parameters and it's meaning: I got these from the solarman pro app, specifically the resource file DYRealTime.txt. There are other files mapping register values, but I cannot verify. If you want to experiment a bit, have a look at the attached zipfile.

Once you know which values apply to you inverter, you can experiment by making a backup of parameters.yaml and constructing a new one with the correct settings. see https://github.com/StephanJoubert/home_assistant_solarman/blob/main/customization.md

[lookup_files.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/8030069/lookup_files.zip)
.

evandrobubiak, 2022-02-18:
@StephanJoubert, is it possible to genarate a device that all the entities belongs to it? For example, generate a Inverter device and all the entities inside it.

StephanJoubert, 2022-03-01:
@evandrobubiak , I think there is a high probability that the micro inverters would be treated as "string inverter", You can try the component with the lookup_file set as deye_string.yaml and then have a look.  Just update to the latest then add the extra setting in the configuration file.

~~~ YAML
 - platform: solarman
    name: . . .
    inverter_host: . . .
    inverter_port:  . . .
    inverter_serial: . . .
    scan_interval:  . . .
    lookup_file: deye_string.yaml
~~~

felipebraz, 2022-03-04:
> The tests you did look very promising. What I would do is to create a small python app to just issue raw requests with the solarman protocol, and then see if I get a response back. Here is a python [...]

Hi, I'm sorry for taking so much time to answer... New born at home, a lot of work,...
I'm trying to do this test, I confess which my knowlogy in Python is less than nothing, I'm a JS, Node,... guy, but I'm trying to learn.

About your plugin I'm doing some debugs but so far I just found some warnings which seems to be a timeout:
2022-03-04 17:33:12 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.deye_pv1_voltage is taking over 10 seconds

I'll keep trying to test what you suggest and any news I inform you.

felipebraz, 2022-03-09:
I´ve tried in so many ways that I can't even count, but I couldn't make it work. Sending "WIFIKIT-214028-READ" through 48899 UDP goes well, but beyond this I'm having some trouble. First because almost every tools that I found are TCP, and my inverter refuses TCP:

[image]

When the subject is network protocols, specially serial, I'am a complete dumb and sometimes I need a little help. Sorry for keep bothering you, I´ve choosed your integration because it is complete and simple to implements, but as usual the things are a little more complicated with me LoL.

Do you have any tips or ideas for helping me through this connection odyssey?

felipebraz, 2022-03-16:
Thank you very much for your attention, patience and dedication!
I've tried to use Docklight Scripting and I realized that I don´t know anything about anything... LoL
I'll stop for now because I'm getting more stressed than results, I really don't know what to do with all of this. But I saved everything you told and I'll try again in a short future, maybe I get more lucky.
My area of knowledge is more web and API develpment, so serial, MODBUS,... is kind of a mistery to me yet, but I'm gonna studdy it soon.