StephanJoubert/home_assistant_solarman #17: No information about Power Production.

jendrasik, 2022-02-09:
Total Production, Total Power, PV2 Power are empty. Other values seems to be OK.

Inverter SN is 21xxxxxxxx
Logger SN is 17xxxxxxx

I am using integration in version 1.1.0

StephanJoubert, 2022-02-09:
@jendrasik ,  Does it show up as "unavailable" empty or zero?

I saw a similar issue on my system where the value showed up as "unavailable", but then found that the reason was that the entity was duplicated (don't know if it is me messing around or the HA update I did this morning). In my case I had two entities sensor.deye_pv2_power which was correct, and sensor.pv2_power_2 which is duplicated.

jendrasik, 2022-02-09:
[image]
it shows 0.

jendrasik, 2022-02-09:
[image]
 Another example of 0. Voltage and Current are show properly.

StephanJoubert, 2022-02-09:
It seems that the register-definition for your inverter is slightly different to mine. What is the make and model of the inverter?

kion666, 2022-02-09:
Hi StephanJoubert and thanks for this nice integration. I have an SofarSolar  and it looks very similar as DEYE. It is now connected to solarman. But I have some problems with integration in HA. Ale the data are unavailable

[image]

Can You support me in that case?

jendrasik, 2022-02-10:
> What is the make and model of the inverter?

Deye SUN-7K-G03
Logger version is: LSW3_15_FFFF_1.0.65

kjacek1, 2022-02-13:
Hello,
I have the same problem
only one string connected - PV1
I see the correct voltage and current for PV1
power is 0V ( it looks like the parameter "sensor.deye_daily_load_consumption" indicates current power - The value of this parameter corresponds to the current production displayed on the inverter)
Moreover, Parametrer sensor.deye_daily_production is 0.

In Home assistant I see the correct value  for [sensor.deye_total_grid_production]

Deye SUN-6K-G03
LSW3_15_FFFF_1.0.72

[image]

StephanJoubert, 2022-02-15:
Hi @jendrasik , @kjacek1 . It seems that other chaps also experienced the same with the 3-phase inverters as discussed on https://github.com/StephanJoubert/home_assistant_solarman/issues/19, and offered to help debug this. Hopefully we will be able to find the correct register-mapping between all.

To enable collaboration, I have also added the ability to specify a lookup_file in the configuration, so just add the lookup_file to the config. We can then try to configure this file correctly for the 3-phase inverter  as described here https://github.com/StephanJoubert/home_assistant_solarman/blob/main/customization.md.

Changes to config file:

~~~ YAML
  - platform: ...
    name: ...
    inverter_host: ...
    inverter_port: ...
    inverter_serial: ...
    scan_interval: ...
    lookup_file: deye_3p.yaml 
~~~

What would also help is to see which parameters are correct, which seems to return values, but where the parameter-names are  incorrect, and which are listed on solarmanpv.com, but not in the integration. This may be useful in searching for the parameter definitions.

kjacek1, 2022-02-15:
Hello,
thank you for the information
I will try to help

I pasted the link below
how did you manage to extract from the application “solarman pro” mobile app
files that contain modbus addresses?

I am a registered user of solarman pro app
I'm reading the data of my inverter but I can't find such data

https://github.com/StephanJoubert/home_assistant_solarman/issues/6#issuecomment-1026941062

StephanJoubert, 2022-02-16:
@kjacek1 Re the parameters in the solarman pro app. If I dig into the app iteself I can extract the "assets" stored inside the apk, and found the lookup files there. Actually saw some others using DYReadlTime.txt and then dug around to see where they got it from.

It is good that you have the solarman pro app working and maybe we can find the real lookups that way. Could you connect to your inverter on "local mode" and then take a screen capture when it shows the correct parameters. We can try to match these up with the  lookup files and see if there is a correlation. Once we have the lookup file, I can easily convert from the text to a lookup yaml file.

See the attached zip file - it is the lookup files I could gather from the solarman pro app, as well as from other sources. I suspected that FSYRealTime.txt may be applicable, but cannot confirm.

[lookup_files.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/8077221/lookup_files.zip)

kjacek1, 2022-02-16:
Option    “ lookup_file: deye_3p.yaml”
does not work for me

I copied the data from the file deye_3p.yaml to parameters.yaml
I limited the scope of the data for me to groups “solar”, “inverter”
My inverter has no battery support,
nor do I have a meter

I am sending what I see in the attachments.

I made 2 changes

- name: "Gen Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      ##scale: 1
      scale: 0.1
      rule: 1
      registers: [0x0054]
      ##registers: [0x00A6]

these records correspond to the current production
Solar man info - AC Output Total Power (Active) 947W

unfortunately the variable after the inverter is switched off in the evening remains at around 300 watts

- name: "Total Production"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 3
      registers: [0x003F,0x0040]
      ##registers: [0x0060,0x0061]
      icon: 'mdi:solar-power'

after such a change the total pruduction is also correct

I'll keep looking

kjacek1, 2022-02-16:
Unfortunately the "local mode" in my solarman app does not want to work.
The application claims that I am giving the wrong SN logger number and I cannot proceed,
(I enter the number either by qr code or by manual entry - the effect is the same)

StephanJoubert, 2022-02-19:
Hi @kjacek1 , @jendrasik Just want to let you know that we have traced the issue to a subtle difference between the register definitions of hybrid and string inverters. Using the new info, I created a file "deye_string.yaml" and pushed it. 

~~~ YAML
sensor:
  - platform: ...
    name: ...
    inverter_host: ...
    inverter_port: ...
    inverter_serial: ...
    scan_interval: ...
    lookup_file: deye_string.yaml
~~~

For now you will have to download or pull from github until I create a release which will trigger HACS to update.

Please have a look and see if you get luck - also feel free to tinker to get it working. For completeness, this is the file I used as reference to create the yaml file [sunsynk_modbus.pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/8102248/sunsynk_modbus.pdf). (The file uses decimal addresses so parameter 73 will be 0x0049)

closed 2022-02-27 by StephanJoubert