StephanJoubert/home_assistant_solarman #18: SofarSolar unknown data in entities

kion666, 2022-02-09:
Hi @StephanJoubert and thanks for this nice integration. I have an SofarSolar  and it looks very similar as DEYE. It is now connected to solarman. But I have some problems with integration in HA. All the data are unknown

[image]

[image]

Can You support me in that case?  Thanks in advance

StephanJoubert, 2022-02-09:
I have created a yaml file from the data I have. It was not possible to test at all so please expect problems :) See attached zipfile.

To test. 
1. Make a copy of parameters.yaml  in the custom_components/solarman folder
2. Overwrite parameters.yaml with the file parameters_sofar.yaml. (Rename parameters_sofar.yaml to parameters.yaml) 
3. See if you get any success, and if so, which parameters need tweaking.

I include the file I used as source to get the parameter definition SOFARMap.txt and the yaml file generated from it that you can use as starting point.

[lookup_files.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/8035192/lookup_files.zip)

kion666, 2022-02-09:
Thank you very much Stephan. I will try to test it tomorrow morning (now my child's are going sleep)  and of course I will let you know. I really appreciate your help.

kion666, 2022-02-10:
Hi @StephanJoubert. I just checked HA and I can see that some of the data are not correct like daily and  total production. I checked the paramaters.yaml and SofarMAE.txt and it looks like the registers are the same.  Some of entities are not neccesary in my inverter, but I cane remove them myself :) But I don't now what to do with wrong daily, total production and  time.
Other parameters are shown correctly like power/voltage.
[image]

Thanks in advance.

grzegorz189, 2022-02-10:
- name: "Total Production"
      class: "energy"
      uom: "kWh"
      scale: 1
      rule: 3
      registers: [0x0016,0x0015]
      icon: 'mdi:solar-power'      

    - name: "Total generation time"
      class: ""
      uom: "h"
      scale: 1
      rule: 3
      registers: [0x0018,0x0017]
      icon: 'mdi:clock-outline'

grzegorz189, 2022-02-10:
[image]

grzegorz189, 2022-02-10:
requests:
  - start: 0x0000
    end:  0x0027
  - start: 0x0105
    end: 0x0114

parameters:
 - group: solar
   items: 
    - name: "PV1 Power"
      class: "power"
      uom: "W"
      scale: 10
      rule: 1
      registers: [0x000A]
      icon: 'mdi:solar-power'

    - name: "PV2 Power"
      class: "power"
      uom: "W"
      scale: 10
      rule: 1
      registers: [0x000B]
      icon: 'mdi:solar-power'

    - name: "PV1 Voltage"
      class: "voltage"
      uom: "V"
      scale: 0.1
      rule: 1 
      registers: [0x0006]
      icon: 'mdi:solar-power'

    - name: "PV2 Voltage"
      class: "voltage"    
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0008]
      icon: 'mdi:solar-power'      

    - name: "PV1 Current"
      class: "current"
      uom: "A"
      scale: 0.01
      rule: 1
      registers: [0x0007]
      icon: 'mdi:solar-power'      

    - name: "PV2 Current"
      class: "current"
      uom: "A"
      scale: 0.01
      rule: 1
      registers: [0x0009]
      icon: 'mdi:solar-power'      

    - name: "Daily Production"
      class: "energy"
      uom: "kWh"
      scale: 0.01
      rule: 1
      registers: [0x0019]
      icon: 'mdi:solar-power'      

    - name: "Total Production"
      class: "energy"
      uom: "kWh"
      scale: 1
      rule: 3
[... 470 lines cut ...]
      -  key: 16
         value: "ID69 Bus voltage is unbalanced and causes unrecoverable fault"
      -  key: 32
         value: "ID70 Grid current is too high and causes unrecoverable fault"
      -  key: 64
         value: "ID65 PV Input Mode Configuration is wrong and causes unrecoverable fault"
      -  key: 128
         value: "ID72 Reserved"
      -  key: 256
         value: "ID73 Reserved"
      -  key: 512
         value: "ID74 Input current is too high and causes unrecoverable fault"
      -  key: 1024
         value: "ID75 Error reading from EEPROM"
      -  key: 2048
         value: "ID76 Error writing to EEPROM"
      -  key: 4096
         value: "ID77 Relay fauilure causes unrecoverable fault"
      -  key: 8192
         value: "ID78 Reserved"
      -  key: 16384
         value: "ID79 Reserved"
      -  key: 32768
         value: "ID80 Reserved"
      icon: 'mdi:wrench'

grzegorz189, 2022-02-15:
Very good idea, "state_class" - ok,
I still have a problem with showing the status of history, you can take a look at it
[image]
Thanks

grzegorz189, 2022-02-15:
I have one more problem after restarting home assistant and when the inverter is turned off, most often at night, all entities are unknown
[image]

kion666, 2022-02-15:
Hi @StephanJoubert  and @grzegorz189 
I will try to check the new config for sofar tomorrow and I will let you know.  Thanks for update state class. Now I have to manually add this to daily and totally production. It will be easier to configure. Best regards

StephanJoubert, 2022-02-16:
@grzegorz189 Yea the entities will do that if home assistant is restarted while the inverter was offline. However, if the entities has values and the inverter goes offline, the values will be maintained.

If you want to use these in automations and it presents a problem, you should be able to look at the status_connection or status_lastUpdate to determine if the values can be used.

kion666, 2022-02-16:
Hi once again @StephanJoubert I confirm that after added sofar yaml to configuration.yaml some of the entities ar unavailable. However the most important for me as daily production, Power, Voltage are working ok.

[image]

Thank You very much for your help.

binoke, 2022-02-16:
I have a 3Phase invertor (Sofar HYD 10KTL-3PH-A).  I have tested the code and integration in HA.  I see that the connection to my invertor works, but the values are unknown or Unavailable.  Would it be possible to create a parameters file for my invertor.   I'm happy to do myself but my modbus knowledge is non-existing.

kion666, 2022-02-16:
@binoke Did You add the sofar_lsw3.yaml to Your config and do You have this file in config\custom_components\solarman?  I also have the KTL Sofar and most important entities are working correctly for me.

binoke, 2022-02-16:
@kion666 What I did was as listed in the original discription:
- Backup original parameters.yaml 
- copy sofar_lsw3.yaml to config\custom_components\solarman
- deleted parameters.yaml and renamed sofar_lsw3.yaml to parameters.yaml 

so when I look at the sensor.  I see it's connected and updating
[image]
[image]

but all other sensor values are Unknown or Unavailable.

kion666, 2022-02-16:
I just set the lookup file as:
[image]
And downloaded all repo to solarman:
[image]

cityslicker99, 2022-02-16:
Hi does this help?
[SOFARSOLAR ModBus-RTU Communication Protocol (1).pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/8079826/SOFARSOLAR.ModBus-RTU.Communication.Protocol.1.pdf)
p

StephanJoubert, 2022-02-16:
Hi @kion666 , that is correct. You can now edit the sofar_lsw3 file to get it working as expected.

kion666, 2022-02-16:
Thanks once again @StephanJoubert   I close the issue and going to buy you coffee :)

closed 2022-02-16 by kion666