StephanJoubert/home_assistant_solarman #20: Instruction Update

McAllstar, 2022-02-16:
This is more an FYI but I noticed in the instructions the config.yaml refers to the lookup_file: solar_lsw2.yaml however when I looked in the solarman directory the file is called solar_lsw**3**.yaml

StephanJoubert, 2022-02-16:
Thanks for that - updated the readme file

closed 2022-02-16 by StephanJoubert

McAllstar, 2022-02-16:
> CAUTION - EXTERNAL SENDER - Please be careful when opening links and attachments. Nedbank - IT Information Security Department (ISD)

You’re a machine, Thanks

Nedbank disclaimer and confidentiality notice:

This email may contain information that is confidential, privileged or otherwise protected from disclosure. If you are not an intended recipient of this email or all or some of the information contained therein, do not duplicate or redistribute it by any means. Please delete it and any attachments and notify the sender that you have received it in error. Unless specifically indicated, this email is neither an offer or a solicitation to buy or sell any securities, investment products or other financial product or service, nor is it an official confirmation of any transaction or an official statement of Nedbank. Any views or opinions presented are solely those of the author and do not necessarily represent those of Nedbank. Nedbank Ltd Reg No 1951/000009/06.

The following link displays the names of the Nedbank Board of Directors and Company Secretary. [http://www.nedbank.co.za/terms/DirectorsNedbank.htm]

If you do not want to click on a link, please type the relevant address in your browser
********************