StephanJoubert/home_assistant_solarman #22: Add support for Sofar Solar 2700 TL inverter, status_connection & reporting 0

zeillemaker, 2022-02-19:
Hi,

To start @StephanJoubert thanks for the integration!

I have a Sofar Solar 2700 TL inverter, the logger has serial number 807xxxxxx and uses the v5 communication protocol. Based on the discussion from issue #18 I made a parameters file. See [link](https://github.com/zeillemaker/home_assistant_solarman/blob/main/custom_components/solarman/sofarsolar_2700tl.yaml). Can support for this inverter be added to the integration?

Separately I have 2 questions/issues:

1. The status_connection parameter does not seem to work properly with my inverter. While all the parameters are captured properly, the status_connection is 'Disconnected' most of the time. I can't seem to figure out why this is reported incorrectly. See attached screenshot.

[image]

2. I use the daily_production parameter in my Home Assistant Energy dashboard. The daily_production is sometimes reported as 0, the data after that is reported correct. This causes Home Assistant Energy dashboard to report the production incorrect. See attached screenshots. I think this is because of the logic behind the use of the state_class total_increasing. See for example the two drops to 0 between 11:00 and 12:00 and the resulting Solar Production from the energy dashboard. All values of the inverter seem the be reported as 0 at those instances. Can the daily_production and other paramaters be prevented from reporting 0?

[image]

[image]

[image]

[image]

StephanJoubert, 2022-02-21:
Hi @zeillemaker , I have added a lookup file for sofar as part of the repo. Is is currently named sofar_lsw3, but the contents are similar to yours, but I can see you made improvements. I will update it as per your settings.

Re the other two questions:
I also see the connection sometimes drop, but in my case it is 99% up and 1% down.  My suspicion is that it is due to the wifi not being perfect (my HA server is on ethernet and dongle is WiFi), however I (and some others) See https://github.com/StephanJoubert/home_assistant_solarman/issues/23 also experience the intermittent invalid parameters, so I will investigate further.

StephanJoubert, 2022-02-27:
@zeillemaker , I have made improvements with regards to the communication and the spikes in data should be gone. Initial tests shows that it made a big difference.

closed 2022-02-27 by StephanJoubert