StephanJoubert/home_assistant_solarman #23: Integration stops  intermittently

ErickvanWyngaard, 2022-02-21:
Hi @StephanJoubert 

Great work on this integration! 

I have a Sunsynk 8kW inverter and have been using it for a few days now and had two issues:
1. Power spikes on PV 2. On the attached image the are spikes to about 30kW on PV2. This didn't happen on PV1 at all.

2. I have lost comms with the inverter a few time. I have circle two times on the attached image. When it first happened I uninstalled the integration and reinstalled it again (that is when it went to 0). Then it started to work again for a shot while without me doing anything, and then it lost comms again.

Do you know what the cause can be?
Is there a way I can get logs to see what the problem is?

Thanks

[image]

StephanJoubert, 2022-02-21:
Hi @ErickvanWyngaard , I also see this issue - maybe less that others but it also irritates me quite a bit. This is next on my list to investigate and hopefully will be in next release.

StephanJoubert, 2022-02-27:
Hi @ErickvanWyngaard , I have made improvements with regards to the communications and tests on my side shows the connection state constant, and the spikes also disappeared. This is in release 1.3.0.

closed 2022-02-27 by StephanJoubert