StephanJoubert/home_assistant_solarman #25: 11KTLX-G with LSW-3 download problem

paku2020, 2022-03-06:
Hi.

Thanks for the integration, great job :)
I wanted to run it in my installation. Yesterday I launched Sofar 11KTLX-G with LSW-3 usb wifi. The logger is in the LSW3_15_270A_1.32 software version.
The problem is that the HA is downloading strange data or not. They are all zeros.
Look:
[image]
or
[image]
See "Installation of PV-SofarSolar Total generation time"
3 932 176 h? :) 
Something is wrong.

The installation works fine and is fine. The inverter does not display errors and is generating electricity.
Does anyone have a similar problem?

greetings

paku2020, 2022-03-17:
Hi,
That's what I defined :)
Today I did the integration again, from scratch.
It looks like:
[image]
[image]
[image]
I have a new version of 11KTLX-G3. Maybe this is the problem?

pelemarse, 2022-03-19:
The same problem. Maybe anyone has an idea how to fix it?

StephanJoubert, 2022-03-23:
Hi @pelemarse @paku2020 , I see that @dsgepers managed to get this right (see issue https://github.com/StephanJoubert/home_assistant_solarman/issues/27 ). Have you managed to solve?

pelemarse, 2022-03-23:
[#27](https://github.com/StephanJoubert/home_assistant_solarman/issues/27) bad configuration ip address. In this case everything is fine with IP, I know registry 0000 is read well, but there are problems with other registries.

paku2020, 2022-03-24:
@StephanJoubert 
Hi,
In my case, the IP address is correct.
Even some data is backed up from the inverter.

i-jk, 2022-09-12:
I have exactly this issue with a new Sofar KTLM-G3 - exactly the same data corruption noted above by @paku2020. IP and other config is correct.

Is there a different data format for newer inverers or loggers?

vytasmk, 2022-10-13:
Totally same problem. 
Inverter: 12KTLX-G3
Logger: LSW3_15_270A_1.53 (Serial: 236xxxxxxx)

Was playing with different registers but no luck. At least it shows something, but not was expected. And there is no information what registers means what. Have found an old (2017) PDF with register table but they not work.

Hope somebody will find solution to this...

nrc04, 2022-11-11:
Hello
try with this registers, should be ok

`requests:
  - start: 0x0404
    end:  0x042E
    mb_functioncode: 0x03
  - start: 0x0484
    end: 0x04BC
    mb_functioncode: 0x03
  - start: 0x0584
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0684
    end: 0x069B
    mb_functioncode: 0x03`

then see here
[SOFAR-G3.External.Modbus.Protocol-Engilsh-V211103.xlsx](https://github.com/StephanJoubert/home_assistant_solarman/files/9987881/SOFAR-G3.External.Modbus.Protocol-Engilsh-V211103.xlsx)

mayor89, 2023-05-05:
Hi. I have LSMW3_15_270A_1.50 and SOFAR 8.8KTLX-G3 and same issue.
Has anyone solved it please?

waaslandwolf, 2023-05-30:
You might want to try this: https://gist.github.com/Floppe/83365483954d8bcee8d8854eeb3fa818

waaslandwolf, 2023-05-30:
I finally got it working with this file: [custom_parameters.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/11604190/custom_parameters.yaml.txt) (with LSMW3_15_270A_1.50 and SOFAR 8.8KTLX-G3)