StephanJoubert/home_assistant_solarman #26: Including Meter readings

vsmarques, 2022-03-07:
Hi  Stephan,
Great work with this integration for Home Assistant !
I have a Sofar Lsw3 Inverter that i've configured at it worked right out of the box. I can access my Inverter data with readings every minute ( Something that Solarman Cloud app can't provide...only 15 mins readings are available, even after customer support changed remotly the upload times to 1 minute).
But i also have a solarman Wifi meter (DDZY422-D2-W) that works in a similar way like the logger for the inverter. I have its serial number, i can query port 8899 and have a response with IP and serial...but i haven't managed yet to get the readings from it, using any of the provided .yaml parameters file.
I have  the document SOFARSOLAR MODBus-RTU communication protocol.pdf, and tried to look for the addresses of the readings that i can read correctly from the inverter to check if they are directly the ones configured in sofar_lsw3.yaml, but they are not the same addresses...
Do you have any clues where to look to find the correct addresses (for parameters and requests start/end ) for this meter ?
Other question that i have : the integration supports more than one device ? can i configure both in configuration.yaml (with different IPs and serials )  ?

Thank you !

StephanJoubert, 2022-03-16:
Hi @vsmarques ,  the parameters depend on the modbus registers of the meter itself (in the case of the inverters I found the info at deye/sofar and not specifically at solarman). I had a look and cannot see any documents, but there is another option. Install the solarman pro app, and see if you can interrogate the meter that way. If you are able to do that, I will look into the source code to see if I can find the parameters.

Would be nice to support that as well

vsmarques, 2022-03-17:
Hi,
I've tried to install solarman Pro App, but it requires a business license for registration, which i don't have. 
I've installed Solarman and Solarman Smart. With those, i can see the meter registered, but i believe they are using the public rest API.
I tried to add the meter to HAcomponent additionally to the inverter logger (defined IP and serial) and it shows as connected in  status. Also updates correctly the lastUpdate time. Only the test value is unknown...but perhaps is just the wrong register address.
[image]

"OTHER" is the meter, and "SOLARMAN" is the default inverter logger.

StephanJoubert, 2022-03-23:
@vsmarques I had a look and cannot find any technical data that lists the modbus registers for this meter. Without that it is extremely difficult to determine the registers. 

You can maybe contact solarman and request the information from them. If they provide you with the info it would be relatively simple to create a yaml file from that.

closed 2022-05-09 by StephanJoubert