StephanJoubert/home_assistant_solarman #27: Over ethernet?

dsgepers, 2022-03-19:
I have a SOFAR 4KTLM-G2. However I'm not connected over WiFi but over wired ethernet. For the whole rest I have access to the whole admin screen and see my IP, serial number etc...

[image]

However I'm only getting as "disconnected" status in HA. 

I'm using the following config in my configuration.yaml:
```
sensor:
  - platform: solarman
    name: SOFAR
    inverter_host: 192.168.0.183
    inverter_port: 8899
    inverter_serial: 2103705955
    scan_interval: 30
    lookup_file: sofar_lsw3.yaml
``` 

However the status remains disconnected and no data is coming in:
[image]

[image]

Can you shed any light on my predicament? :-)

I DO have a solution where I just call the solarmanpv.com API online through an API key i requested, however your solution with direct access is just so much more elegant imho.

pelemarse, 2022-03-19:
@dsgepers Is the IP entered correctly?

Wireless STA mode IP - 192.168.1.183
Config  IP - inverter_host: 192.168.0.183

dsgepers, 2022-03-19:
ffs...i really am a special kind of idiot...

closed 2022-03-19 by dsgepers

dsgepers, 2022-03-19:
Btw, it now provides me with the correct info! :-) Thanks!

sanderbaks, 2023-05-07:
@dsgepers Hi Dennis, (I think fellow Dutch?)

I still can not get it to work :(. Is it still working for you ?

Thanks!

Sander