StephanJoubert/home_assistant_solarman #29: Energy data for Micro Inverter Setup

martinisonline, 2022-03-22:
Hello,

First, many thanks for this great peace of software.

I have a 5KW Deye inverter and everything works fine with the code reading implementation.

I have a micro inverter connected to the GEN port to reuse some old installed equipment.
I noticed that all the energy being produced by the micro inverter in the GEN port, is not being counted into the total PV production Energy counters.

Downloaded an export from solarman smart web, and the value of energy produced daily and total by the micro inverter GEN port is available in the excel file as "Daily Production MI(kWh)" and "Total Production MI(kWh)".

Do you know in each register these values are stored and if is it possible to get this data?

I assume that my Total Energy Production would be the sum of the PV production + Micro Inverter production.

Thank you.

StephanJoubert, 2022-03-23:
Hi @martinisonline , I can unfortunately not confirm, but would suspect that the value for the entity "sensor.deye_micro_inverter_power" would contain the power. However, the unit is set as Watt, and is probably instantaneous power (not kWH). It should be possible to use this value in a template to sum it, but it would not be trivial.

The modbus registers are described in the attached document - I see only the register already exposed in the document:
[sunsynk_modbus.pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/8334292/sunsynk_modbus.pdf)

If you manage to do this, you should be able to add this as an additional production to the energy dashboard so it can do the summation for you.

Hope you have success.

StephanJoubert, 2022-04-09:
Hi @martinisonline , Thanks for the effort. I was a bit busy the last few weeks and only catch up now.

 I had another look at the document (even translated the Chinese) and confirm that the doc does not include the two values. Can you may test by modifying the parameters.yaml file, adding the two values.

To add then, just edit the file with a text editor (I can recomend using VS Code) and add the extra parameters as follows:

``` YAML
    - name: "Microinverter Daily Energy"
      class: "energy"
      state_class: "measurement"
      uom: "kWh"
      scale: 1
      rule: 1
      registers: [0x003E]
      icon: 'mdi:solar-power'

    - name: "Microinverter Total Power"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 1
      rule: 1
      registers: [0x005C]
      icon: 'mdi:solar-power'
```

This is assuming that the 62, and 92 you referred to was the decimal values for which the hex values are 3E and 5C respectively . You can play around with the scale if it needs to be scaled before display.

If this works, please send me the final details and I will include in the distribution, of if you are familiar with git, do a "pull request"

andregoncalvespires, 2022-04-13:
Hi.

Maybe I did something wrong, but trying to this configuration, it doesn't worked. 

I have four micro-inverters 2KW, model SUN-2000G3 4 MPPT 220V DEYE. If anybody can help integrate them, I will be very grateful.

douglasfl, 2022-04-18:
> Hi.
> 
> Maybe I did something wrong, but trying to this configuration, it doesn't worked.
> 
> I have four micro-inverters 2KW, model SUN-2000G3 4 MPPT 220V DEYE. If anybody can help integrate them, I will be very grateful.

Hi,
I have an inverse of this same model.
I integrated mine using it as:
lookup_file: deye_string.yaml

andregoncalvespires, 2022-04-19:
Thank you! Great!

chipsi, 2022-05-15:
https://github.com/StephanJoubert/home_assistant_solarman/issues/46 here is microinverter config.

closed 2022-06-16 by StephanJoubert