StephanJoubert/home_assistant_solarman #32: status_connection = Disconnected

mrp-yt, 2022-03-27:
Got everything setup. Everything is identical compared to the manual except Device serial Number and firmware. Mine is  LSW3_14_FFFF_1.0.38 where in the screenshot inside manual is LSW3_14_FFFF_1.0.29.
 
Once i get all setup and added all sensors to dashboard everything showing "unknown" and next to Status_Connection i get "Disconnected" 

Can someone please guide me to the right direction how to get this going?

StephanJoubert, 2022-03-30:
@mrp-yt , I would say that the most probable cause would be an incorrect IP address, or a dynamic IP. To check , see if you can connect to port 8899 on the IP provided. There are a few tools to do that, you could even use telnet to confirm.

If that is not the case, the logger may be using the version 4 of the specification. The serial no should suffice to verify that.

mrp-yt, 2022-03-30:
> @mrp-yt , I would say that the most probable cause would be an incorrect IP address, or a dynamic IP. To check , see if you can connect to port 8899 on the IP provided. There are a few tools to do [...]

Thank you for reply!

**About IP** : I am using same IP which allows me to get in to dashboard. 
http://192.168.178.53/index_cn.html
[image]

**Serial Number**:  SM1ES136K7X941

piposant, 2022-04-01:
I have the same problem.
My firmware of Device is version LSW3_15_FFFF_1.0.72. The serial number start with 17xxxc

HeartandHana, 2022-04-01:
I found the same problem as the comments above. Please advise me to downgrade protocol to v4. 
Appreciate your support.

StephanJoubert, 2022-04-02:
@HeartandHana I found an integration for  Omnik that used the older protocol, but from what I can see the modbus registers are hard-coded so it may work for the inverter, but it will be a long shot. There are also some integrations that gets the info from the solarman portal (for which you have to request an "API key") from what I can see.

https://github.com/hultenvp/home_assistant_omnik_solar

StephanJoubert, 2022-04-02:
@mrp-yt  @piposant from the firmware versions it seems as if it should work. Could you post he inverter make and model and the configuration you are using? @mrp-yt , the serial no SM1ES136K7X941 looks strange, it should be the collector's serial no and not the inverter serial no and expect something beginning with 17XXXXXX . Can you confirm?

mrp-yt, 2022-04-02:
@StephanJoubert 
I have Sofar Hybrid Inverter 3-6kW. 
And yes, you are correct. I took wrong serial number. Correct number : 170527XXXX

piposant, 2022-04-03:
I have a Deye SUN 5K, and user de configuration: lookup_file: deye_string.yaml

StephanJoubert, 2022-04-04:
@piposant , From what I can see that is the hybrid inverter that this integration used as starting point. This is the default, so to test, just remove the line specifying the lookup file.

piposant, 2022-04-04:
Removed the line and still the same problem. Disconnected.

piposant, 2022-04-04:
Sorry, I found the problem and it was my mistake with the IP... It already connects.

mrp-yt, 2022-04-09:
Spent some time trying all sorts of settings. Have no luck at all to get this thing working.......

Not sure at this moment if its even worth it ...

StephanJoubert, 2022-04-13:
@paqp , it is correct to leave the inverter_mb_slave out. The documentation on github is a little ahead of the releases and we will make a new release soon. With regards to the configuration, just make doubly sure of the IP and the serial no of the logger (I see that your post has it unpopulated).

(Note that is is the serial no of the logger and not the inverter).

paqp, 2022-04-13:
@StephanJoubert  I edited my post removing serial just because it's not wise leave serials on internet... xD  
Anyway, I put wifi device serial, I tried the inverter too (just to try it) but is alphanumeric so Home Assistant doesn't accept it.
I'm still disconnected

paqp, 2022-04-21:
@StephanJoubert  this is the log from home assistant.
I don't know if could help.

`2022-04-21 21:56:43 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030000002845d43e15
2022-04-21 21:56:45 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51000101500fb4b16c06a0201b6a40a08982000004612575a0500f015
2022-04-21 21:56:45 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030000002845d43e15
2022-04-21 21:56:47 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51000101500fc4b16c06a0201b8a40a089a2000004612575a0500f515
2022-04-21 21:56:47 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030105001055fb6315
2022-04-21 21:56:49 DEBUG (SyncWorker_2) [custom_components.solarman.solarman] a51000101500fd4b16c06a0201baa40a089c2000004612575a0500fa15
2022-04-21 21:57:43 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030000002845d43e15
2022-04-21 21:57:45 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a51000101500fe4b16c06a0201f2a40a08d42000004612575a05006b15
2022-04-21 21:57:45 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030000002845d43e15
2022-04-21 21:57:47 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a51000101500ff4b16c06a0201f4a40a08d62000004612575a05007015
2022-04-21 21:57:47 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030105001055fb6315
2022-04-21 21:57:49 DEBUG (SyncWorker_7) [custom_components.solarman.solarman] a51000101500004b16c06a0201f6a40a08d82000004612575a05007515
2022-04-21 21:58:43 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51700104500004b16c06a02000000000000000000000000000001030000002845d43e15
2022-04-21 21:58:45 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] a51000101500034b16c06a02012ea50a08102100004612575a0500ea15`

mgarcia0, 2022-04-23:
Hi,
I think I have the same issue. The connection status appears as Disconnected, but if I enable the debug, I think there is connection with the logger, and data is received:

2022-04-23 16:45:40 DEBUG (SyncWorker_3) [custom_components.solarman.solarman] a51000101500217f76196a0201a6af0600f60000006e615d620500b515

May be that data is received, but cannot be parsed correctly? My inverter is Sofar-HYD, but I have not been able to get this working with any of the lookup files.
Any ideas?

Thanks in advance

mgarcia0, 2022-04-27:
Hi,
I have just sorted it out. It turns out my inverter has different registers than the ones provided. I have a Solar HYD inverter, whose register definition are available here:

https://community.openenergymonitor.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf

I have adapted the inverter definition and now it reads all the info. Attached the file if anyone find it useful.
Remove .txt and leave it with .yaml extension.

Best regards,

[Sofar_Hyd_lsw3.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/8577653/Sofar_Hyd_lsw3.yaml.txt)

paqp, 2022-04-27:
BOOOM BABY! It worked!
I have a branded ZCS inverter, and I can read lots of data. I think this yaml file should be refined and integrated in a future upgrade of this plugin!

Well done! Thank you @mgarcia0  and @StephanJoubert

paulhardman, 2022-07-18:
> Hi, I have just sorted it out. It turns out my inverter has different registers than the ones provided. I have a Solar HYD inverter, whose register definition are available here: [...]

Just to give another +1 to this YAML - I couldn't get my HYD-ES to talk to HA Solarman properly, but this was exactly the answer. Thank you so much @mgarcia0 !

rjpearce, 2022-07-28:
> Hi, I have just sorted it out. It turns out my inverter has different registers than the ones provided. I have a Solar HYD inverter, whose register definition are available here: [...]

Thanks for this, can you link the thread you got it from as I can't find it.

The fix doesn't work for me (HYD6000EP) so I may need to make a few more tweaks, do you have an HYD ES?

Jochey, 2022-08-03:
Hi, I am having the same issue using an LSW-3 Stick logger together with an INVT iMars inverter. I am able to connect via http and also use telnet via putty. I am receiving logs in home assistant, so I am quite sure the communication is running.

2022-08-03 21:43:21.656 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a51700104500007883cc660200000000000000000000000000000203022a001c64408c15
2022-08-03 21:43:23.661 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a51000101500517883cc66020105430000ce120000d68cea6205009115
2022-08-03 21:43:23.666 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a51700104500007883cc660200000000000000000000000000000203022a001c64408c15
2022-08-03 21:43:25.671 DEBUG (SyncWorker_1) [custom_components.solarman.solarman] a51000101500527883cc66020107430000d0120000d68cea6205009615

I have tried all the different inverter definitions.

Is there anyone having any experience with the same inverter setup or have some tips to check out?

cyberrep, 2022-08-18:
> * SOLARE

did you have correct values for yours?
mine just wrong values

meteosat007, 2023-01-08:
Hi All, Just killed the HACS installation, removed all files and restarted HA. Next Cloned the repo and copied the Solarman folder into custom_components on the HA instace which is the OS version. Restarted HA then configured the confog flow with my details for a Sofar HYD3600-ES using the appropriate .yaml file and everything has loaded as expected with reqular updates at the same time as scrapping the data collection into an Influxdb before sending the data collected to the Solarman servers as described using the [serck4 ](https://github.com/serek4/node-red-sofar-inverter) github collection.

So using the HACS method seems to fail with the older code, but the clone of the latest works fine.
Hope this helps you all....

fluccats, 2023-08-26:
Hi Folks, I believe that my product needs protocol V4. Please how can I configure this?