StephanJoubert/home_assistant_solarman #33: Sometime Output Active Power is read out with the value 0

vebse, 2022-04-02:
I am really happy that there is this integration for the Homeassistant! Great thing and thanks a lot!!!
I have a Solarmax inverter with the Datalogger Stick LSW-3. So far everything works great - except for one thing.
Again and again the Output Active Power of the inverter is read out with the value 0. See also the screenshot. What could be done about it? I have increased the readout period from 30 to 60 seconds. But this does not bring any improvement. 

[image]

Greetings
Markus

StephanJoubert, 2022-04-02:
Hi Markus, I also found this with the DEYE inverter and suspected that we may interpret no data as a valid packet. To counter that I added all the CRC checks to validate the response before using, but still get it sometimes. This leads me to think that that the logger outputs the data that way. I have an idea how to try and solve and will let you know once we have a solution.

Leny1996, 2022-06-22:
Hi, I have the same issue on one Sofar inverter. I have two sofar devices connected. Looks like first one works fine, as on screenshot below.
[image]
For the same period on second device I get:
[image]
Wifi connection on both is 50-55%, so connection is kinda stable.
First one is on V270, second on V230. Maybe it's firmware issue, but I'm scared of updating the second one tbh.