StephanJoubert/home_assistant_solarman #39: Home Assistant reports problem with state class

dagwud, 2022-04-12:
Thanks for this fantastic integration - soo nice having nearer real time data!

From logs:
```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:312 
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22)) 
First occurred: 12:25:10 AM (1 occurrences) 
Last logged: 12:25:10 AM

Entity sensor.inverter_total_grid_production from integration solarman has state class total_increasing, but its state is not strictly increasing. Triggered by state 5148.2 (5148.6) with last_updated set to 2022-04-11T22:23:29.807720+00:00. Please report it to the custom component author.

StephanJoubert, 2022-04-17:
@dagwud Thanks for that. The parameter should be increasing only, but I will have a look and see if it happens on my side as well.

dagwud, 2022-04-17:
Yeah it didn’t make much sense to me either that total grid production could decrease. 

If there’s any more info I can provide just shout

closed 2022-05-09 by StephanJoubert