StephanJoubert/home_assistant_solarman #40: HA setup issues

oaktek, 2022-04-16:
Hi @StephanJoubert, thanks so much for this integration! Wondering if you can help me as I cant figure this out and I have followed your instructions.

Firstly in configuration.yaml i have the following:

sensor:
  - platform: solarman
    name: DEYE 
    inverter_host: 192.168.2.2
    inverter_port: 8899
    inverter_serial: 23339##### ---- I haven taken out the rest of the serial for this post, newer version dont know if this is the problem 
    #inverter_mb_slaveid: 1 ------- had to comment this out otherwise the integration wont start
    scan_interval: 30
    lookup_file: deye_string.yaml

Then in the energy configuration I do not have the option you show in your screenshots, this is what i have:
[image]

Then when i do choose some options i have errors saying that last_reset is missing and I cant add the battery info:
[image]

This is what my energy dashboard looks like:
[image]

This is the sensor info i have:
[image]
[image]
[image]

I would so appreciate your help!

oaktek, 2022-04-16:
Hi, sorry found my main issue! I had to use the parameters.yaml and now im getting more info. Can you tell me if im missing anything or if anything is wrong?

Heres the list I now get in energy configuration:

[image]
Last one being: Deye Total Production

[image]
In the solar production im still getting that one paramter missing as you can see and in the batteries area i dont have the options you have

Below the energy dashboard and something doest seem right, you will see a photo of the inverter display
[image]
[image]
So you will see the solar production on Kw, the house usage and whats coming in from eskom (power company) doesnt seem right, also not too sure about the battery info

pepelillo, 2022-04-16:
Not sure if this can help, but make sure that you put your CT clamp in the proper way (see doc)

StephanJoubert, 2022-04-17:
@oaktek , I think you may have an older version of Home Assistant that requires the "last-reset" parameter. I had it in initially, but with the update they recommend that it be taken out. The dashboard is not 100% the same as the inverter screen since the Home Assistant dashboard shows a cumulative icon, and the inverter show it in real-time.

Let the dashboard collect info for a day and then see if the results look correct.

midlakelands, 2022-04-18:
Hi,
I was having similar issues with the solis-hybrid parameters, after reading [the thread](https://community.home-assistant.io/t/home-energy-management-in-2021-8/325539/157) on it I decided to change some of the daily parameters but I did it with the following in configuration.yaml as that allowed me to experiment without changing the original files.  After an hour the stats started to work (I believe the energy card updates at 12 mins past each hour) 

homeassistant:
  customize:
    # Add an entry for each entity that you want to overwrite.
    sensor.solarman_daily_generation:
      state_class: total_increasing
    sensor.solarman_today_battery_discharge:
      state_class: total_increasing
    sensor.solarman_today_battery_charge:
      state_class: total_increasing

dsgepers, 2022-04-20:
I've acctually been solving this by copying the configuration and adding an additional field:

``` 
    - name: "Daily Production"
      class: "energy"
      state_class: "total"      
      uom: "kWh"
      scale: 0.01
      rule: 1
      registers: [0x0019]
      icon: 'mdi:solar-power'   

    - name: "Daily Production (Total Increasing)"
      class: "energy"
      state_class: "total_increasing"      
      uom: "kWh"
      scale: 0.01
      rule: 1
      registers: [0x0019]
      icon: 'mdi:solar-power' 
``` 

Which is less than perfect, because this morning (after updating last night), no solar generation showed up in the dash and the custom property was gone. I think I might give @midlakelands solutions a try, that seems solid.

CreateYourAccount-username, 2022-04-26:
I have the same issue with a Solis inverter. All the other things work fine, it's just adding it to the Energy dashboard that doesn't work due to the "Last reset missing
The following entities have state class 'measurement' but 'last_reset' is missing:
sensor.solarman_daily_generation" error.

I'm using solarman 1.4.1 and HA 2022.4.7

midlakelands, 2022-04-27:
> I have the same issue with a Solis inverter. All the other things work fine, it's just adding it to the Energy dashboard that doesn't work due to the "Last reset missing The following entities have [...]

Try the solution i gave above (replacing your sensor name for daily generation and battery charge/discharge) it'll update the energy card in an hour or so. Alternatively you can update the original yaml file but I didn't bother change it. You will need to either restart HA or reload the customization yaml to see those updates.

The explanation for why I changed them to total_increasing is below
[image]

CreateYourAccount-username, 2022-04-27:
> Try the solution i gave above (replacing your sensor name for daily generation and battery charge/discharge) it'll update the energy card in an hour or so. Alternatively you can update the original [...]

I have put those duplicate sensors in the solarman solis_hybrid yaml. I think it's working okay, slowly getting some results in. I'll see how it performs in the sunshine tomorrow. Thanks!

closed 2022-05-09 by StephanJoubert