StephanJoubert/home_assistant_solarman #42: config-flow support broken

jmccrohan, 2022-04-17:
Hi Stephan,

Thanks for pulling the 1.4.0 release together.

Just to let you know that eab8fb0 seems to have broken support for both new and existing config-flow configurations.

Cheers,
Jon

StephanJoubert, 2022-04-18:
Hi Jon, thanks for this - I'm looking to see if I can find the problem.

StephanJoubert, 2022-04-18:
Found and fixed the issue (there was an "await" missing ).  
``` python
async def async_setup_entry(hass: HomeAssistant, entry: ConfigEntry, async_add_entities: AddEntitiesCallback):
    _LOGGER.debug(f'sensor.py:async_setup_entry: {entry.options}') 
    await async_setup_platform(hass, entry.options, async_add_entities)
```
Instead of one calling the other, I made both call a new (synchronous) method. Should be fixed in 1.4.1

closed 2022-04-18 by StephanJoubert