StephanJoubert/home_assistant_solarman #43: Deye hybrid 3 phase

gfuentesv, 2022-04-30:
Hello,
I need profile or documentation for integrate a 3 phase Deye hybrid inverter.

Thankls.

skupi20, 2022-08-15:
I created custom parameters for my deye hybrid 3 phase inverter. Its working, i based on [https://github.com/maxx-ukoo/deye-modbus2mqtt/blob/master/docs/Modbus%20amit%20posun.docx](url)

```
requests:
  - start: 520
    end: 525
    mb_functioncode: 0x03
  - start: 529
    end: 529
    mb_functioncode: 0x03
  - start: 534
    end: 535
    mb_functioncode: 0x03
  - start: 598
    end: 600
    mb_functioncode: 0x03
  - start: 616
    end: 619
    mb_functioncode: 0x03
  - start: 625
    end: 625
    mb_functioncode: 0x03
  - start: 636
    end: 636
    mb_functioncode: 0x03
  - start: 653
    end: 653
    mb_functioncode: 0x03
  - start: 672
    end: 679
    mb_functioncode: 0x03

parameters:
 - group: solar
   items:
    - name: "PV1 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [672]
      icon: 'mdi:solar-power'

    - name: "PV2 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [673]
      icon: 'mdi:solar-power'

    - name: "PV1 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [676]
      icon: 'mdi:solar-power'

    - name: "PV2 Voltage"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [678]
      icon: 'mdi:solar-power'

    - name: "PV1 Current"
      class: "current"  
      state_class: "measurement"
      uom: "A"
      scale: 0.1
      rule: 1
      registers: [677]
      icon: 'mdi:solar-power'

    - name: "PV2 Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
      scale: 0.1
      rule: 1
      registers: [679]
      icon: 'mdi:solar-power'

    - name: "Daily Production"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 1
      registers: [529]
      icon: 'mdi:solar-power'

    - name: "Total Production"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 3
      registers: [534,535]
      icon: 'mdi:solar-power'

    - name: "Total Inverter Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [636]
      icon: 'mdi:solar-power'

 - group: grid
   items:
     - name: "Total Grid Power"
       class: "power"
       state_class: "measurement"
       uom: "W"
       scale: 1
       rule: 2
       registers: [625]
       icon: 'mdi:transmission-tower'

     - name: "L1 out of Grid Power"
       class: "power"
       state_class: "measurement"
       uom: "W"
       scale: 1
       rule: 2
       registers: [616]
       icon: 'mdi:transmission-tower'

     - name: "L2 out of Grid Power"
       class: "power" 
       state_class: "measurement"
       uom: "W"
       scale: 1
       rule: 2 
       registers: [617]
       icon: 'mdi:transmission-tower'

     - name: "L3 out of Grid Power"
       class: "power" 
       state_class: "measurement"
       uom: "W"
       scale: 1
       rule: 2 
       registers: [618]
       icon: 'mdi:transmission-tower'

     - name: "Total out of Grid Power"
       class: "power"
       state_class: "measurement"
       uom: "W"
       scale: 1
       rule: 2
       registers: [619]
       icon: 'mdi:transmission-tower'

     - name: "Grid Voltage L1"
       class: "voltage"
       state_class: "measurement"
       uom: "V"
       scale: 0.1
       rule: 1
       registers: [598]
       icon: 'mdi:transmission-tower'

     - name: "Grid Voltage L2"
       class: "voltage"
       state_class: "measurement"
       uom: "V"
       scale: 0.1
       rule: 1
       registers: [599]
       icon: 'mdi:transmission-tower'

     - name: "Grid Voltage L3"
       class: "voltage"
       state_class: "measurement"
       uom: "V"
       scale: 0.1
       rule: 1
       registers: [600]
       icon: 'mdi:transmission-tower'

     - name: "Daily Energy Bought"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1
       rule: 1
       registers: [520]
       icon: 'mdi:transmission-tower-export'

     - name: "Total Energy Bought"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1
       rule: 1
       registers: [522,523]
       icon: 'mdi:transmission-tower-export'

     - name: "Daily Energy Sold"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1
       rule: 1
       registers: [521]
       icon: 'mdi:transmission-tower-import'

     - name: "Total Energy Sold"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1
       rule: 3
       registers: [524,525]
       icon: 'mdi:transmission-tower-import'

 - group: load
   items:
     - name: "Total Load Power"
       class: "power"
       state_class: "measurement"
       uom: "W"
       scale: 1
       rule: 1
       registers: [653]
       icon: 'mdi:lightning-bolt-outline'

```

nesati, 2022-09-01:
> I created custom parameters for my deye hybrid 3 phase inverter. Its working, i based on [Modbus amit posun.docx](https://github.com/maxx-ukoo/deye-modbus2mqtt/blob/master/docs/Modbus%20amit%20posun.docx)

Thanks a lot for the config file. Thanks to the linked document a some trial and error I managed to figure out the registers for battery information. So here they are. Hope someone will find them useful.

```
 - group: battery
   items:
     - name: "Today charge of the battery"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1
       rule: 1
       registers: [514]
       icon: 'mdi:battery-arrow-up'
     - name: "Today discharge of the battery"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1 
       rule: 1 
       registers: [515]
       icon: 'mdi:battery-arrow-down'
     - name: "Battery capacity"
       class: "power_factor"
       state_class: "measurement"
       uom: "%"
       scale: 1  
       rule: 1  
       registers: [588]
       icon: 'mdi:battery-outline'
     - name: "Total charge of the battery"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1  
       rule: 1 
       registers: [516]
       icon: 'mdi:battery-arrow-up'  
     - name: "Total discharge of the battery"
       class: "energy"
       state_class: "total_increasing"
       uom: "kWh"
       scale: 0.1  
       rule: 1  
       registers: [518]
       icon: 'mdi:battery-arrow-down'
     - name: "Battery voltage"
       class: "voltage"
       state_class: "measurement"
       uom: "V"
       scale: 0.01
       rule: 1
       registers: [587]
       icon: 'mdi:battery-outline'
```

Note that for this to work you also need to append 

```
  - start: 514
    end: 518
    mb_functioncode: 0x03
  - start: 587
    end: 588
    mb_functioncode: 0x03
```
to the `requests` section.