StephanJoubert/home_assistant_solarman #44: Last reset missing

toledobastos, 2022-05-06:
The sensors are working flawlessly, but the energy panel can’t display them correctly because they don’t seem to reset at midnight. This is the error message I get from the Energy panel:

The following entities have state class 'measurement' but 'last_reset' is missing:
sensor.solis_local_daily_generation

The reported figures are therefore cumulative instead of daily. Am I missing something here? The plots in your example show the Energy tab working correctly.

michaljonik, 2022-06-03:
I had the same issue. I was getting a negative spike every midnight. 
Also my custom calculations of energy bought and sold were messy - most likely because of very less frequent updates during nights when there was is production.
So I ended up encapsulating _solarman_total_production_ into a utility meter and with this new entity it works perfectly:
```
utility_meter:
  daily_energy_generated:
    source: sensor.solarman_total_production
    name: Daily energy production
    cycle: daily
```

viking2010, 2022-06-13:
I had the same issue but resolved it a different way. I'm using a Solis Inverter and the solis_hybrid.yaml file, but the same should apply for the other makes yaml files.....all you need to do is find the "daily generation" in the file and change the state_class from measurement to total_increasing as below. Once done, restart HA and those warnings should be gone. Hope that helps!

` - group: Generation
   items: 
    - name: "Daily Generation"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 1
      registers: [33035]
      icon: 'mdi:solar-power'`

PR #57 (2022-06-16): solis_hybrid.yaml: Multiple fixes
* Optimise Modbus requests
* Fix Daily Generation state_class (Fixes #44)
* Fix Monthly Generation register typo (Fixes #56)
commit 6819111b (2022-06-16): solis_hybrid.yaml: Multiple fixes

* Optimise Modbus requests
* Fix Daily Generation state_class (Fixes #44)
* Fix Monthly Generation register typo (Fixes #56)
closed 2022-06-16 by StephanJoubert