StephanJoubert/home_assistant_solarman #45: HYD3000 / HYD6000 protocol

paqp, 2022-05-08:
Hi! I started from your files to map my HYD6000. 
After weeks of researches I mapped a 77 components of inverter!
My model is branded "ZCS AZZURRO" but I saw that's the same for unbranded ones.
And this is what I got:
[image]
[image]
[image]

A lot of datas!
This inverter has NO separate entities to show charging/discharging batteries or giving/getting energy from grid, but it has one entity for grid in/out and one for batteries charge/discharge. 
So if, for example, entity  "batteries charge/discharge power" has a positive value, batteries are charging, if it has negative value they are discharging.
So I'm thinking how I could use this to integrate with HA because it requires 2 distinct entities for grid and for batteries.

I hope it will help
[hyd3000-6000.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/8646417/hyd3000-6000.txt)

Cheers!
Paq

samdunne, 2022-05-09:
Hey Paq. I am using the same inverter and have done more or less the same as what you have above. Have you had problems with massive spikes in your data?

paqpal, 2022-05-09:
yes, more or less... I'm using filter in HA.
Are you using the "daily" parameters in the Energy configuration for grid and batteries?

samdunne, 2022-05-09:
I am. Would you be able to share your filter and config for the energy dashboard?

paqp, 2022-05-09:
This is my configuration:

```
  - platform: filter
    name: "Pannelli Produzione Istantanea - filtrato"
    entity_id: sensor.solarman_pv_instant_generated_pw
    filters:
      - filter: range
        lower_bound: 0.0
        upper_bound: 10.0

  - platform: filter
    name: "Battery Percentage - filtered"
    entity_id: sensor.solarman_battery_percentage
    filters:
      - filter: range
        lower_bound: 0.0
        upper_bound: 100.0

  - platform: integration
    source: sensor.pannelli_produzione_istantanea_filtrato
    name: energy_spent
    unit_prefix: k
    round: 2
```
I use the last one `- platform: integration` for instant generated power conversion from kW to kWh.
    

[image]

It's not perfect but it's working. I'm changing something everyday.
This is the last file I made.
[HYD3000-6000 parameters-v2.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/8651774/HYD3000-6000.parameters-v2.txt)

StephanJoubert, 2022-06-16:
Thanks @paqp . I have pulled into the main branch and it will be in the next release. If you want to make changes, just clone the repo and make a pull request.

closed 2022-06-16 by StephanJoubert