StephanJoubert/home_assistant_solarman #46: Microinverter DEYE SUN2000G3 (new config)

chipsi, 2022-05-10:
Hello.
I made a new YAML file for the new DEYE / VESDAS SUN2000G3 microinverter. It has 4x MPPT and 4x inverter. This is only the first version. Maybe it will help someone.
Thank you to the author for a great job and other beautiful energy day.
ChipSi007

.txt rename to .yaml
[deye_4mppt.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/8664097/deye_4mppt.txt)

tcwed-br, 2022-05-29:
> Hello. I made a new YAML file for the new DEYE / VESDAS SUN2000G3 microinverter. It has 4x MPPT and 4x inverter. This is only the first version. Maybe it will help someone. Thank you to the author [...]

Hello Stephan / Chipsi 
great job  4pcs DeyeSun1600 now working in HA ... trying to figure out how to get the "radiator temperature" out of the box ...
many thanks and keep on going

chipsi, 2022-05-30:
Hello.
I am trying it too, but it return zero. I'll let you know when I find it.

chipsi, 2022-05-30:
I have got it. Add this

      - name: "Radiator temperature"
        class: "temperature"
        state_class: "measurement"
        uom: "°C"
        scale: 0.01
        rule: 1
        registers: [0x005A]
        icon: "mdi:thermometer"

tcwed-br, 2022-06-02:
Hello 
will need to copy 4mppt.yaml to the correct path 
and use the device serial number ( not inverter serial ) 
config.yaml 
platform: solarman  
name: DEYExyz
inverter_host: 192.168.xx.yy
inverter_port: 8899  
inverter_serial: 4041xxxxx
inverter_mb_slaveid: 1  
lookup_file: deye_4mppt.yaml
...
ps radiator temperature works thanks chipsi

wmjv, 2022-06-02:
Thanks! It worked!

StephanJoubert, 2022-06-16:
Hi @chipsi. Thanks for the good work. I have pulled onto the main branch and it will be added to the next release of the component. Keep well!

closed 2022-06-16 by StephanJoubert

alessandroo23, 2022-08-23:
Chipsi, Leonardolisboa, Tcwed-br, Thanks! It worked. data is now being collected.

VaillantHassIo, 2022-11-03:
this works for Bosswerk MI600 ( OEM ) as well

giovanne123, 2022-11-03:
for at least Bosswerk MI600 you can also add for Daily/Total per Modul (example is for the 2 Mppt of the Bosswerk). 
Using it on my HA/Solarman intergration already
```

- name: "Daily Production PV1"
class: "energy"
state_class: "total"
uom: "kWh"
scale: 0.1
rule: 1
registers: [0x0041]
icon: 'mdi:solar-power'

- name: "Daily Production PV2"
class: "energy"
state_class: "total"
uom: "kWh"
scale: 0.1
rule: 1
registers: [0x0042]
icon: 'mdi:solar-power'

- name: "Total Production PV1"
class: "energy"
state_class: "total_increasing"
uom: "kWh"
scale: 0.1
rule: 3
registers: [0x0045,0x0046]
icon: 'mdi:solar-power'

- name: "Total Production PV2"
class: "energy"
state_class: "total_increasing"
uom: "kWh"
scale: 0.1
rule: 3
registers: [0x0047,0x0048]
icon: 'mdi:solar-power'
```