StephanJoubert/home_assistant_solarman #48: New Config for Solis 1P8K-5G

gedger, 2022-05-12:
May help someone else.
Renamed from YAML to allow upload.
[solis_1p8k-5g.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/8675742/solis_1p8k-5g.txt)

goul, 2022-05-22:
Worked fine for my S5-GR1P(2.5-6)K - thank you.

gedger, 2022-05-30:
First step is to search the net to see if you can find a Solis register specifications for your inverter. If you're really lucky you may find yours. Solis has been known to send them out if you ask but I'm still waiting....However, they give you a good idea of what to look out for. Try searching "Solis RS485_MODBUS Communication Protocol" in the first instance, I found a couple that were close but none were exact so I did the following:-

Get pysolarmanv5 working from the command line. [https://github.com/jmccrohan/pysolarmanv5](url)

Use the provided example "register_scan.py" to find all the registers that your inverter has. This is just a matter of starting at zero and working up but if my config file mostly works then I suggest starting with 2999-3120 (input registers) and 2999-3305 (holding registers).

Once you have the registers you then need to identify the values. I started a spreadsheet and did several scans and identified the registers that changed every scan. These were obviously the real-time data outputs, I then matched the values up with those reported on SolisCloud plus tried to match up with the specs I found. Bear in mind that many of them need multiplying to get the real value and some of them span more than one registers, i.e. high and low part of a long integer.

Good luck!!