StephanJoubert/home_assistant_solarman #53: Issue with logger 23xxxx

Loudo42, 2022-06-02:
Hi,

I have an inverter Sofar HYD 6000 EP. This model is quiet new and I have a logger 23xxxxx with the firmware LSW3_15_270A_1.32.
I tried the set up but the inverter is not connected and I have unknown values for entities.
I have HA on a RPI arm64 with RaspberryPIOS. 
At this moment, I have only tested integration with the automatic way and selected the sofar_lsw3.yaml look up file . I will try the manual one but it will certainly not change
I did not found the v4 protocol which is recommend for logger different of 17xxx and 21xxx, could you please let me know where I can find it? I am sorry I try to find it but it escapes me.
Nevertheless maybe my logger is not compatible with v4 protocol too.
So do you have a solution.
I have also found a solution with Sofar2mqtt but it is more time-consumming and complicated, and can it do my goal describes below.

My goal is to have realtime data from the inverter to manage the charge of my Tesla and also to pilot the inverter for charging and discharging schedule and/or max power output.
If piloting is not possible, does not matter, my main goal is to have realtime data for the EV charge.

Thank you very much in advance for your help.

Regards,
Loudo

gedger, 2022-06-05:
May be worth posting the front web page of your inverter which shows the  version serial etc. of your system. Also post up any output from. http://YOURIPADDR/config_hide.html

There seems to be a huge variation of logger versions and firmware out there plus a lot of old stock. Have you tried contacting the manufacturer, some are more helpful than others.

igieon, 2022-06-06:
I have new hyd 5kw i am testing new yaml setting its here but my inverted died after 2 days of working so i will continue when i get new ;-/just rename it to add aditional register and production i will add later on
[sofar_g3hyd.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/8843037/sofar_g3hyd.yaml.txt)
.

santo76, 2022-06-24:
i have HYD 6000 EP with 23XXXXX and i can't receive data from datalogger.
any soluiton for this?

thanks

Loudo42, 2022-06-26:
Hi,

Igieon, I have tried your config files, I have 34 entities (41 with the original one), but still no value or connection.

My logger firmware is LSW3_15_270A_1.53
For my inverter 
- the software version of the communication CPU is 030012
- the sofware main version is 030011
- Hardware version 001

Any ideas?

Thank you.

Floppe, 2022-08-24:
Hello!

I'm in the same boat. Got my Sofar Solar 8.8KTLX-G3 inverter up and running yesterday. Tested this integration and every sensor have no data. My logger has the same firmware, and same serial prefix.

SW: 000001
HW: 101

Any info I could assist with for finding the issue. Perhaps a tcpdump between HA and the logger on port 8899? Here below only the actual data communication:

HA -> Logger port 8899
00000000  a5 17 00 10 45 00 00 3f  af 92 8c 02 00 00 00 00
00000010  00 00 00 00 00 00 00 00  00 00 01 03 00 00 00 28
00000020  45 d4 bf 15

Logger -> HA
00000000  a5 13 00 10 15 00 57 3f  af 92 8c 02 01 01 43 00
00000010  00 ea 26 00 00 99 7c 05  63 01 90 02 00 03 05 15

HA logs says only: 
`WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solarman_pv1_power is taking over 10 seconds`

Thanks.

cyberrep, 2022-09-01:
same here, I cant get any correct date
Inverter serial number 21xxxxx
Device serial number 23xxxx
Firmware version LSW3_15_FFFF_1.0.78

Txominr, 2022-09-11:
Me too same problem
Unknown status, no data
Inverter serial number 21xxxxx
Device serial number 23xxxx
Firmware version LSW3_15_FFFF_1.0.78

i-jk, 2022-09-12:
I have this too -- seems #25 is also this issue.

otello86, 2022-10-12:
Same issue here with Sofar inverter HYD-5000ES and logger :

S/N : 234XXXXXXXX
Firmware : LSW3_15_FFFF_1.0.65

steve-cox, 2022-10-15:
Not sure why, but my Sofar HYD-6000ES works (mostly).....

S/N: 234xxxxxxx
Firmware: LSW3_15_FFFF_1.0.65

Using pysolarmanv5, I found that the inverter/logger responds to the normal WIFIKIT discovery packet, and that it responds to modbus holding register reads on 0x0200-0x0255 and input register reads at 0x10B0-0x10BC & 0x2000-0x200f.

This means that the sofar_wifikit profile is the closest matching one.  The missing PV information is at 0x252-0x254, and there are a handful of other changes elsewhere (e.g. no additional grid supplies).

Using the sofar_wifikit profile as the base, I have created a new profile that I believe is mostly correct - I am unsure if battery cycles & emergency V/A are in the correct location and I have really no idea how to test the battery type is correct without cracking the case open.  I've attached the profile in a zip file:

[sofar_hyd_x000.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/9792153/sofar_hyd_x000.zip)

However.... I did say it 'mostly' works.  Either there is a bug on the code, or the inverter has a habit of sending back corrupted values every now and then.  Sometimes my panels are generating 250kW(!), sometimes 1kW... at 2am.  Other parameters are also affected, mostly dropping to zero.  With debug turned on (and additional debug added to print the Exception), I can even see strings fail to parse (incorrect number of characters for struct.unpack)

I'll keep looking into it, but hopefully the profile I've added to this message helps some out.

steve-cox, 2022-10-15:
@Floppe : From what I've discovered when HA complains about solarman taking too long, it's because the inverter/logger doesn't respond to requests for modbus registers that are not present.

ie. in my above comment, I mentioned that my HYD 6000 responds to requests for holding register reads on 0x0200-0x0255 and input register reads at 0x10B0-0x10BC & 0x2000-0x200F.  If I made a request outside of that range, there would be a delay.

Solarman tries twice for each request row in the yaml.  Each request has a 10s timeout (solarman.py::send_request: "sock.settimeout(10)").  As such there can be a considerable delay when the wrong registers are polled.

I'm not sure if the above yaml profile will help you as you appear to have a different inverter.  If it doesn't, the following Python will help you locate the registers available:

```
from pysolarmanv5 import PySolarmanV5, V5FrameError
import umodbus.exceptions

def main():    
    
    modbus = PySolarmanV5("<my inverter/logger ip>", <my logger serial 234xxxxxxx>, port=8899, mb_slave_id=1, verbose=True)

	#
	# This scan will take a LONG time!
	#
	
    print("Scanning holding registers - use mb_functioncode: 0x03 in YAML request section")
    for x in range(0x0000, 0xffff, 0x100):
        try:
            val = modbus.read_holding_registers(register_addr=x, quantity=1)[0]
            print(f"Register: {x:05x}\t\tValue: {val:05} ({val:#06x})")
        except (V5FrameError, umodbus.exceptions.IllegalDataAddressError):
           # print(f"Register: {x:05x}\t\tValue: NONE")
            continue
    print("Finished scanning holding registers")

    print("Scanning input registers - use mb_functioncode: 0x04 in YAML request section")
    for x in range(0x0000, 0xffff, 0x100):
        try:
            val = modbus.read_input_registers(register_addr=x, quantity=1)[0]
            print(f"Register: {x:05}\t\tValue: {val:05} ({val:#06x})")
        except (V5FrameError, umodbus.exceptions.IllegalDataAddressError):
            continue
    print("Finished scanning input registers")

if __name__ == "__main__":
    main()
```

It will then be a case of making reads of all of those registers at the same time a 5 minute update appears into the solarmanpv.com, and matching them up (taking into account the scaling).  The following code will read and print the registers for my HYD 6000 - you can modify this for yours:

```
    modbus = PySolarmanV5("192.168.16.8", 2343467077, port=8899, mb_slave_id=1, verbose=True)
    try:
        # Read 86 register values starting at 0x200 (ie: 0x200-0x255 inclusive)
        retval = modbus.read_holding_registers(0x0200, 86)
        print(retval)

        # Read 12 register values starting at 0x10B0 (ie: 0x10B0-0x10BC inclusive - my inverter 
        # actually responds up to 0x10C7, but I have no idea what those values are for!)
        retval = modbus.read_input_registers(0x10B0, 12)
        print(retval)

        # Read 16 register values starting at 0x2000 (ie: 0x2000-0x200F inclusive)
        retval = modbus.read_input_registers(0x2000, 16)
        print(retval)
        print(try_parse_ascii(retval))
    except (V5FrameError, umodbus.exceptions.IllegalDataAddressError):
        print('Nope')

```
Occasionally, my logger returns values with invalid checksums - wait a second, then try again.  This may be related to the corrupted values problem I mention in my previous post.

otello86, 2022-10-17:
Thanks @steve-cox
It seems to be working with your profile file except some datas stay in "available" and others which have incoherent values
I will check that

steve-cox, 2022-10-17:
@otello86 There seems to be a second problem, other than just the definition file - the integration appears to be accepting and parsing modbus messages that are either invalid or of a different type.  Unfortunately, this causes corrupted values to be accepted.

I created a new [bug](https://github.com/StephanJoubert/home_assistant_solarman/issues/120) for it, and will try to investigate further.  Looking forward to getting this integration working with my inverter :)

Floppe, 2022-10-19:
@steve-cox Thank you!

You got me into the right track. Registers seems to be the same as @igieon has in his file, some small modifications as my inverter is only on-grid. Took me a while to figure out why HomeAssistant did not get any values, but the Python script did work flawlessly. Seems the solarman configuration file is very sensitive. One small thing wrong and no data at all is updated to HA although the registers are read OK according to debug logging. Perhaps in need of better error catching and logging there.

Here is what I ended up with: [sofar-g3.yaml](https://gist.github.com/Floppe/83365483954d8bcee8d8854eeb3fa818)

Many thanks to you both!

vytasmk, 2022-10-19:
@Floppe happy for you. I have tried with Your file on my SofarSolar 12KTLX-G3 but no luck all parameters stays "Unknown" :) Can some body point me how to debug communication that is going from HA to Inverter?

steve-cox, 2022-10-19:
@vytasmk add the following to your configuration.yaml, and it will turn on debugging for this integration:

```
logger:
  default: info
  logs:
    custom_components.solarman: debug
```

The raw contents of the packets will now be dumped into the log.  The protocol is documented over at the [pysolarmanv5](https://pysolarmanv5.readthedocs.io/en/latest/solarmanv5_protocol.html) library.

pskuza, 2022-10-25:
I have a SOFAR 15KTLX-G3 Inverter with following logger:

S/N: 234xxxxxxx
Firmware: LSW3_15_270A_1.50

Neither the [sofar-g3.yaml](https://gist.github.com/Floppe/83365483954d8bcee8d8854eeb3fa818) or the merged [sofar_g3hyd.yaml](https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/sofar_g3hyd.yaml) return any usable data.

Did anyone manage to make this integration work with this Inverter? I checked the logs with debug on:

`2022-10-25 13:53:52.349 WARNING (MainThread) [homeassistant.helpers.entity] Update of sensor.solarman_sofarsolar_pv_generation_today is taking over 10 seconds`

and do requests being sent and answered.

chimj, 2022-11-19:
> 

Hi, I'm trying to find out my reg location with your script (LSW3_15_1216_1.06 / 23xxx ) 

For the holding registers I got this 

Scanning holding registers - use mb_functioncode: 0x03 in YAML request section
Register: 00c80         Value: 08258 (0x2042)
Register: 00c90         Value: 04613 (0x1205)
Register: 00ca0         Value: 00000 (0x0000)
Finished scanning holding registers

My next step would be like below? (I tried but keep getting a 'nope' ;-)

  # Read 86 register values starting at 0x200 (ie: 0x200-0x255 inclusive)
        retval = modbus.read_holding_registers(0x2042, 10)
        print(retval)

Thanks.

udivankin, 2023-01-28:
Same here with ZCS Azzurro 3PH 3.3KTL-V3 (rebranded Sofar 6.6KTLX-G3) - neither config works, although the inverter does respond in the example provided by pysolarmanv5, which is promising.

Would like to contribute, but don't know where to start.

maxtoor93, 2023-05-01:
I am in the same case with a HYD-5000-EP
since some time someone has developed a file that works?

frsantos, 2023-05-02:
There is a new inverter definition for KStar inverters that was defined against a 235xxx inverter, don't know if it will work for you: 

https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/kstar_hybrid.yml

The inverter is in the code but not yet released, as far as I know.

otello86, 2023-05-11:
@maxtoor93 i use this file definition with my HYD-5000ES and it works fine
[custom_parameters.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/11452551/custom_parameters.zip)

mayor89, 2023-05-17:
can anyone help me please?
i have same inverter as floppe. used same yaml file but getting this error:

2023-05-17 18:44:51.822 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] invalid modbus crc
2023-05-17 18:44:51.823 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [1028 - 1070] failed, invalid response packet.
2023-05-17 18:44:51.823 DEBUG (SyncWorker_8) [custom_components.solarman.solarman] Querying [1028 - 1070] failed, [0] retry attempts left
2023-05-17 18:44:51.824 WARNING (SyncWorker_8) [custom_components.solarman.solarman] Querying registers [1028 - 1070] failed, aborting.

I wonder: Is it problem that i am already using app on android to get the readings? it says only one device can connect but in settings you can see 2 servers are available..
thank you in advance!

waaslandwolf, 2023-05-30:
I finally got it working with this file: [custom_parameters.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/11604190/custom_parameters.yaml.txt) (with LSMW3_15_270A_1.50 and SOFAR 8.8KTLX-G3)

sanderbaks, 2023-05-30:
Current status: Still connection unknown

mayor89, 2023-05-31:
> I finally got it working with this file: [custom_parameters.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/11604190/custom_parameters.yaml.txt) (with LSMW3_15_270A_1.50 and SOFAR 8.8KTLX-G3)

@sanderbaks :
Superb. Glad to hear that. I have still the same issue. Maybe my logger is not working correctly. Can you please share more details? Maybe logs? Have you changed yaml configuration or are you using everything as it was on installation? Can you screenshot your settings if you changed something? Thank you in advance.

sanderbaks, 2023-05-31:
My status is still on disconnected.. outside of HA everything is working fine, and  can connect to it.

oziee, 2023-06-05:
> > I finally got it working with this file: [custom_parameters.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/11604190/custom_parameters.yaml.txt) (with LSMW3_15_270A_1.50 and SOFAR 8.8KTLX-G3)
> 
> Did you just go in to Studio Code Editor and add the "custom_parameters.yaml" file and copy in the contents? Is it that easy?

this worked for me too.. but my 8KTLM-G3 has 3x PVs mppt's.. anyone know how to add the 3rd PV info to the config?? PV1 and PV2 are working, but PV3 isn't listed

mayor89, 2023-06-05:
> this worked for me too.. but my 8KTLM-G3 has 3x PVs mppt's.. anyone know how to add the 3rd PV info to the config?? PV1 and PV2 are working, but PV3 isn't listed

Hi. I saw an issue where this was discussed (in this git). Please check. I think you just need to edit the config file and add it manually with correct address (I would do it by trying next address). Dont forget to restart after every try and also dont forget to wait few minutes..

oziee, 2023-06-05:
yup the next registered worked perfect

ricoroodenburg, 2023-06-11:
Same issue, tried every suggestion without any luck.

Serialnumber 238x
Firmware version LSW3_15_1E03_1.31

wlgallesio, 2023-09-09:
> sim, o próximo registrado funcionou perfeitamente

Hello, could you help me, how was your pv3?
I tried that and it didn't work
      - name: 'PV3 Power'
        class: 'power'
        state_class: 'measurement'
        uom: 'W'
        scale: 10
        rule: 1
        registers: [0x0592]
        icon: 'mdi:solar-power'

      - name: 'PV3 Voltage'
        class: 'voltage'
        state_class: 'measurement'
        uom: 'V'
        scale: 0.1
        rule: 1
        registers: [0x0590]
        icon: 'mdi:solar-power'

      - name: 'PV3 Current'
        class: 'current'
        state_class: 'measurement'
        uom: 'A'
        scale: 0.01
        rule: 1
        registers: [0x0591]
        icon: 'mdi:solar-power'

wlgallesio, 2023-09-09:
> > > > Finalmente consegui trabalhar com este arquivo: [custom_parameters.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/11604190/custom_parameters.yaml.txt) (com LSMW3_15_270A_1.50 e SOFAR 8.8KTLX-G3)
> > > 
> > > 
> > > Você acabou de entrar no Studio Code Editor e adicionar o arquivo "custom_parameters.yaml" e copiar o conteúdo? É tão fácil?
> > 
> > 
> > isso funcionou para mim também .. mas meu 8KTLM-G3 tem 3x PVs mppt's .. alguém sabe como adicionar as informações do terceiro PV à configuração ?? PV1 e PV2 estão funcionando, mas PV3 não está listado
> 
> Oi. Eu vi um problema onde isso foi discutido (neste git). Por favor, verifique. Acho que você só precisa editar o arquivo de configuração e adicioná-lo manualmente com o endereço correto (eu faria isso tentando o próximo endereço). Não se esqueça de reiniciar após cada tentativa e também não se esqueça de esperar alguns minutos.

Would you know how to do PV3? thanks

oziee, 2023-09-09:
Mine setup is

```

# Sofar 3.3-12KTLX-G3
# Note: no modbus energy meter connected

requests:
  - start: 0x0400
    end:  0x042E
    mb_functioncode: 0x03
  - start: 0x0480
    end: 0x04AF
    mb_functioncode: 0x03
  - start: 0x0580
    end: 0x0592
    mb_functioncode: 0x03
  - start: 0x0680
    end: 0x069B
    mb_functioncode: 0x03
parameters:
  - group: InverterDC
    items:
      - name: "PV1 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 10
        rule: 1
        registers: [0x0586]
        icon: 'mdi:solar-power'

      - name: "PV2 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 10
        rule: 1
        registers: [0x0589]
        icon: 'mdi:solar-power'
        
      - name: "PV3 Power"
        class: "power"
        state_class: "measurement"
        uom: "W"
        scale: 10
        rule: 1
        registers: [0x0592]
        icon: 'mdi:solar-power'

      - name: "PV1 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x0584]
        icon: 'mdi:alpha-v-circle-outline'

      - name: "PV2 Voltage"
        class: "voltage"
        state_class: "measurement"
        uom: "V"
        scale: 0.1
        rule: 1
        registers: [0x0587]
        icon: 'mdi:alpha-v-circle-outline'
    
      - name: "PV3 Voltage"
        class: "voltage"
[... 260 lines cut ...]
             value: "ID21 Leakage current sampling error (DC side)"
          -  key: 32
             value: "ID22 Leakage current sampling error (AC side)"
          -  key: 64
             value: "ID23 Load voltage DC component sampling error"
          -  key: 128
             value: "ID24 DC input current sampling error"
          -  key: 256
             value: "ID25 DC component sampling error of grid current (DC side)"
          -  key: 512
             value: "ID26 DC input branch current sampling error"
          -  key: 1024
             value: "ID27"
          -  key: 2048
             value: "ID28"
          -  key: 4096
             value: "ID29 Leakage current consistency error"
          -  key: 8192
             value: "ID30 Grid voltage consistency error"
          -  key: 16384
             value: "ID31 DCI consistency error"
          -  key: 32768
             value: "ID32"

```

gmcinalli, 2023-11-15:
> @maxtoor93 i use this file definition with my HYD-5000ES and it works fine [custom_parameters.zip](https://github.com/StephanJoubert/home_assistant_solarman/files/11452551/custom_parameters.zip)

Thank you so much, I've the same inverter!

cecheto, 2024-06-01:
I have a Sofar PowerAll 6Kw (ESI 3-6K-S1-HA1-6) with a LSW3 2395XXXXX and firmware LSW3_15_270A_1.68. I finally found that zcs_azzurro-ktl-v3.yaml did work for me and added some more params from [sofar_g3_lsw3_logger_reader](https://github.com/kubaceg/sofar_g3_lsw3_logger_reader) making a custom_parameters.yaml. The new params are for the batteries status.
I share this because it took me a while of searching and testing to make it work.

trezit, 2024-06-10:
> I have a Sofar PowerAll 6Kw (ESI 3-6K-S1-HA1-6) with a LSW3 2395XXXXX and firmware LSW3_15_270A_1.68. I finally found that zcs_azzurro-ktl-v3.yaml did work for me and added some more params from [...]

Thanks a lot @cecheto I was struggling to configure my new SOFAR ESI 6K S1 and the zcs_azzurro-ktl-v3.yaml seems to be working well.
It is possible for you to share the custom_parameters.yaml that you create ? 
Thanks in advance.

andrew-tsunami, 2024-09-14:
@cecheto Hi, I've just installed the ESI 6K S1 and managed to get some data using zcs_azzurro-ktl-v3.yaml per your advice!
Do you mind sharing your experience with getting the battery states available too?

cecheto, 2024-09-16:
> > I have a Sofar PowerAll 6Kw (ESI 3-6K-S1-HA1-6) with a LSW3 2395XXXXX and firmware LSW3_15_270A_1.68. I finally found that zcs_azzurro-ktl-v3.yaml did work for me and added some more params from [sofar_g3_lsw3_logger_reader](https://github.com/kubaceg/sofar_g3_lsw3_logger_reader) making a custom_parameters.yaml. The new params are for the batteries status. I share this because it took me a while of searching and testing to make it work.
> 
> Thanks a lot @cecheto I was struggling to configure my new SOFAR ESI 6K S1 and the zcs_azzurro-ktl-v3.yaml seems to be working well. Is it possible for you to share your custom_parameters.yaml ? Thanks in advance.

Sorry, I did not see you comment. Here is my custom_parameters.yaml:

`# ZCS Azzurro 3-phase non-hybrid inverters
# with LSW-3 WiFi logger with SN 23xxxxxxxx and FW LSW3_15_270A_1.53:
# 3PH 3.3KTL-V3
# 3PH 4.4KTL-V3
# 3PH 5.5KTL-V3
# 3PH 6.6KTL-V3

# Not tested, but could probably work:
# ZCS Azzurro 3PH 8.8KTL-V3
# ZCS Azzurro 3PH 11KTL-V3
# ZCS Azzurro 3PH 12KTL-V3
# SOFAR Solar 4.4KTLX-G3
# SOFAR Solar 5.5KTLX-G3
# SOFAR Solar 6.6KTLX-G3
# SOFAR Solar 8.8KTLX-G3
# SOFAR Solar 11KTLX-G3
# SOFAR Solar 12KTLX-G3

requests:
  - start: 0x0400
    end: 0x042B
    mb_functioncode: 0x03
  - start: 0x0482
    end: 0x04A4
    mb_functioncode: 0x03
  - start: 0x0582
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0600
    end: 0x611  
    mb_functioncode: 0x03
  - start: 0x0682
    end: 0x068B
    mb_functioncode: 0x03

parameters:
  - group: Battery
    items:
      - name: 'Bat1 Voltage'
        class: 'voltage'
        state_class: 'measurement'
        uom: 'V'
        scale: 0.1
        rule: 1
        registers: [0x0604]
        icon: 'mdi:battery-charging'

      - name: 'Bat1 Current'
        class: 'current'
        state_class: 'measurement'
        uom: 'A'
        scale: 0.01
        rule: 1
        registers: [0x0605]
        icon: 'mdi:battery-charging'

      - name: 'Bat1 Power'
        class: 'power'
        state_class: 'measurement'
        uom: 'kW'
        scale: 0.01
        rule: 1
        registers: [0x0606]
        icon: 'mdi:battery-charging'
[... 898 lines cut ...]
          - key: 32
            value: 'ID182 BMS short circuit alarm'
          - key: 64
            value: 'ID183 BMS version inconsistency'
          - key: 128
            value: 'ID184 BMS CAN version inconsistency'
          - key: 256
            value: 'ID185 BMS CAN version is too low'
          - key: 512
            value: 'ID186 '
          - key: 1024
            value: 'ID187 '
          - key: 2048
            value: 'ID188 '
          - key: 4096
            value: 'ID189 Arc device communication failure'
          - key: 8192
            value: 'ID190 DC arc alarm fault'
          - key: 16384
            value: 'ID191 PID repair failed'
          - key: 32768
            value: 'ID192 PLC module heartbeat loss'
`

[custom_parameters.yaml.txt](https://github.com/user-attachments/files/17013116/custom_parameters.yaml.txt)

cecheto, 2024-09-16:
> Thanks a lot @cecheto I was struggling to configure my new SOFAR ESI 6K S1 and the zcs_azzurro-ktl-v3.yaml seems to be working well. Is it possible for you to share your custom_parameters.yaml ? [...]

Sorry I didn't see your message. Here is the custo_parameters.yaml (as txt): 
[custom_parameters.yaml.txt](https://github.com/user-attachments/files/17013154/custom_parameters.yaml.txt)

```
# ZCS Azzurro 3-phase non-hybrid inverters
# with LSW-3 WiFi logger with SN 23xxxxxxxx and FW LSW3_15_270A_1.53:
# 3PH 3.3KTL-V3
# 3PH 4.4KTL-V3
# 3PH 5.5KTL-V3
# 3PH 6.6KTL-V3

# Not tested, but could probably work:
# ZCS Azzurro 3PH 8.8KTL-V3
# ZCS Azzurro 3PH 11KTL-V3
# ZCS Azzurro 3PH 12KTL-V3
# SOFAR Solar 4.4KTLX-G3
# SOFAR Solar 5.5KTLX-G3
# SOFAR Solar 6.6KTLX-G3
# SOFAR Solar 8.8KTLX-G3
# SOFAR Solar 11KTLX-G3
# SOFAR Solar 12KTLX-G3

requests:
  - start: 0x0400
    end: 0x042B
    mb_functioncode: 0x03
  - start: 0x0482
    end: 0x04A4
    mb_functioncode: 0x03
  - start: 0x0582
    end: 0x0589
    mb_functioncode: 0x03
  - start: 0x0600
    end: 0x611  
    mb_functioncode: 0x03
  - start: 0x0682
    end: 0x068B
    mb_functioncode: 0x03

parameters:
  - group: Battery
    items:
      - name: 'Bat1 Voltage'
        class: 'voltage'
        state_class: 'measurement'
        uom: 'V'
        scale: 0.1
        rule: 1
        registers: [0x0604]
        icon: 'mdi:battery-charging'

      - name: 'Bat1 Current'
        class: 'current'
        state_class: 'measurement'
        uom: 'A'
        scale: 0.01
        rule: 1
        registers: [0x0605]
        icon: 'mdi:battery-charging'

      - name: 'Bat1 Power'
        class: 'power'
        state_class: 'measurement'
        uom: 'kW'
        scale: 0.01
        rule: 1
        registers: [0x0606]
        icon: 'mdi:battery-charging'
[... 896 lines cut ...]
          - key: 16
            value: 'ID181 BMS charge/discharge overload alarm'
          - key: 32
            value: 'ID182 BMS short circuit alarm'
          - key: 64
            value: 'ID183 BMS version inconsistency'
          - key: 128
            value: 'ID184 BMS CAN version inconsistency'
          - key: 256
            value: 'ID185 BMS CAN version is too low'
          - key: 512
            value: 'ID186 '
          - key: 1024
            value: 'ID187 '
          - key: 2048
            value: 'ID188 '
          - key: 4096
            value: 'ID189 Arc device communication failure'
          - key: 8192
            value: 'ID190 DC arc alarm fault'
          - key: 16384
            value: 'ID191 PID repair failed'
          - key: 32768
            value: 'ID192 PLC module heartbeat loss'
```

cecheto, 2024-09-16:
> @cecheto Hi, I've just installed the ESI 6K S1 and managed to get some data using zcs_azzurro-ktl-v3.yaml per your advice! Do you mind sharing your experience with getting the battery states available too? I've tried to look into the sofar_g3_lsw3 but found no trace of battery entities.

See previous message.