StephanJoubert/home_assistant_solarman #54: Stop polling when Inverter off-line at night

gedger, 2022-06-05:
When the inverter goes into standby at night the integration continues to poll and generates a large number of HA log messages.

Logger: homeassistant.helpers.entity
Source: helpers/entity.py:746 
First occurred: 4 June 2022, 22:02:32 (458 occurrences) 
Last logged: 05:40:03

Update of sensor.solis_inverter_status is taking over 10 seconds
Update of sensor.electric_consumption_today is taking over 10 seconds
Update of sensor.electric_cost_today is taking over 10 seconds

This is summer when the nights are short and 458 log entries are made, it will be much worse in the winter. Simple fix would be to only log the message once but keep polling. Better fix would be to log the message once and then reduce the polling rate until a response is obtained.

igieon, 2022-06-06:
I think its better to have config which entity is update and when because for example if you have hybrid system you want monitor other entittie like battery discharge and so on ...

gedger, 2022-07-22:
As the daylight duration at my site decreases I'm steadily getting more warning messages so I have created a fix to only poll the Inverter from sunrise to sunset. I've added a check box onto the configuration form to enable/disable this mode for those who need it. Defaults to off so will not effect existing installs.

Before I submit as a push is this likely to be be accepted as a change? You can review the changes on my fork.

https://github.com/gedger/home_assistant_solarman

PS. This is my first go at modifying a HA integration so feel free to point out any issues....but it does work for me!

hajo62, 2023-01-09:
> I have created a fix

@gedger : Isn't there a way, that you can merge your stuff to this branch?
The latest code from here has some useful invalidation rules and yours would keep the logs clean(er).

For me it looks like @StephanJoubert who created this great integration isn't maintaining this anymore?

gedger, 2023-01-10:
In progress, I have merged my changes in with the latest code. This is not my forte so had a few issues merging with git but I think I'm there, there's a new branch in my fork that has the merged changes. Currently being tested on my system but I'm seeing some issues with the config screen....

gedger, 2023-01-24:
I'm struggling to spend any time on this but there is an easier solution which is to disable polling and then set up an automation to poll the parameters you need at the rate you require. Here's my automation as an example. This polls every 30 seconds and starts 10 minutes after sunrise and stops 5 minutes before sunset.
```
alias: PV - Poll Inverter
description: "polls solarman only during daylight"
trigger:
  - platform: sun
    event: sunrise
    offset: "00:10:00"
condition: []
action:
  - repeat:
      while:
        - condition: sun
          before: sunset
          after: sunrise
          before_offset: "-00:05:00"
      sequence:
        - service: homeassistant.update_entity
          data: {}
          target:
            entity_id:
              - sensor.solis_operating_status
              - sensor.solis_status_lastupdate
              - sensor.solis_inverter_current
              - sensor.solis_inverter_voltage
              - sensor.solis_inverter_ac_power
              - sensor.solis_inverter_frequency
              - sensor.solis_inverter_temperature
              - sensor.solis_pv1_current
              - sensor.solis_pv1_voltage
              - sensor.solis_pv2_current
              - sensor.solis_pv2_voltage
              - sensor.solis_total_dc_power
              - sensor.solis_daily_generation
        - delay:
            hours: 0
            minutes: 0
            seconds: 30
            milliseconds: 0
mode: single
```

stonie81, 2023-01-24:
Hi @gedger could you please tell me how to disable polling. Set the scan_interval to 0 or how is it done?

gedger, 2023-01-24:
Easy, go to the integrations page, click on the three dots on the solarman integration, then system options. note, you need HA 21.6 or later.

[https://www.home-assistant.io/blog/2021/06/02/release-20216/](url)

stonie81, 2023-01-24:
Thanks! Should switch off localization 😅
I try using the device_tracker of my fritz.box as trigger and condition, should also work :-)