StephanJoubert/home_assistant_solarman #62: Minor issue: Battery Status incorrect

spencerw67, 2022-06-23:
Great integration - it really helps to monitor my new Deye installation.

Minor issue: Battery Status is always shown as "Charge" despite the inverter showing another status.

qfinucane, 2023-03-27:
I’m having the same issue. I’ve removed the lookup in the code for monitoring and the value for battery status (register 189) remains at 0 for some reason, no matter if the battery is charging, discharging or in idle.

qfinucane, 2023-04-15:
According to the Sunsynk Modbus register manual, register 189 {0x00BD} is for "PV4 input power". There does not appear to be a "Battery Status" register according to the manual.

I use current (Register 191 {0x00BF}) direction to determine the status. In Grafana I have a threshold configured to display "Discharging" if the current is positive, "Charging" if the current is negative and Idle if its under 1 amp.

[sunsynk_modbus - no chinese.docx](https://github.com/StephanJoubert/home_assistant_solarman/files/11240317/sunsynk_modbus.-.no.chinese.docx)