StephanJoubert/home_assistant_solarman #63: Solis S3-WIFI-ST (Gen 3) Data Logging Stick

OneScuzzie2k, 2022-06-25:
Dont seem to be able to use this stick? as its a a 14 digit serial number with letters and numbers, any suggestions?

Ghostofapacket, 2022-06-27:
Resolving this to "work" is a case of changing the end of line 30 on "config_flow.py" from )): int, to )): str,

However I am currently in the same boat and having done this, I am now testing to see if this can even poll the inverter correctly.

Ghostofapacket, 2022-06-29:
Agreed, I did some more work on this and it is due to the hex encoding where it is expecting an int but we're now passing it a string.

I need to spend some time to see how the developer has been doing this request and even if this request is possible on the newer model of loggers. 

I have one here, I will do some looking over the coming days and report back

NickSutton, 2022-07-17:
> Agreed, I did some more work on this and it is due to the hex encoding where it is expecting an int but we're now passing it a string.
>
> I need to spend some time to see how the developer has been [...]

Thank you for your work on this. There are quite a few people at HA Community in the same boat!

morts1, 2022-07-17:
Thank you also for your great work on this. I really hope a way forward can be found with this issue as it seems all new installations are having this new logger.

t0nyb, 2022-08-12:
I have a week old installation in the UK and have the new stick. I cant even get in with admin:admin even though it obviously has a web page and I can ping it.

blongden, 2022-08-21:
The username is admin and the password is your wifi password - however, port 8899 is not open on the S3-WIFI-ST stick so none of these Solarman integration libraries will work. I've not yet found anyone that's managed to read / write directly to the modbus port via this stick so it may not be possible. I also have not yet been able to find a source for the solarman based data loggers that DO expose modbus.

NickSutton, 2022-08-21:
> The username is admin and the password is your wifi password - however, port 8899 is not open on the S3-WIFI-ST stick so none of these Solarman integration libraries will work. I've not yet found [...]

Ended up replacing the stick with a Pi Zero and modbus adapter, reading values perfectly now, just need to work out how to write them

blongden, 2022-08-22:
@NickSutton Looking at doing exactly the same thing so I can talk to the modbus interface over wifi. How did you hook up the modbus adapter to the pi? Can you power the pi from the +5V pin on modbus?

NickSutton, 2022-08-22:
> @NickSutton Looking at doing exactly the same thing so I can talk to the modbus interface over wifi. How did you hook up the modbus adapter to the pi? Can you power the pi from the +5V pin on modbus?

I used this [https://github.com/incub77/solis2mqtt](url) with this adapter [https://www.amazon.co.uk/gp/product/B016IG6X7I/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1](url), soldered it all together and it even fits in the original SOLIS Wifi stick case. Powered by the 5V pin from the solis

dezfutak, 2022-09-17:
> Ended up replacing the stick with a Pi Zero and modbus adapter, reading values perfectly now, just need to work out how to write them

Any joy in being able to write?

NickSutton, 2022-09-17:
> 

Nope. I was thinking to find a Python guy on Fiverr or something to see if a freelancer can make it happen

jonathanfurness, 2023-02-19:
Wondering if there is any update on this thread? I self-installed the Solis Hybrid 5G inverter with a Solis Datalogger WiFi stick (v3) and as others have described, won't work with Solarman. 

Nick, as a Pi person, would love to find out more. Have you documented this any where?

J

dezfutak, 2023-02-21:
I found a Gist which enables direct communication via an RS485 to USB converter connected to the COM output on the Solis (so no Solis dongle) & have it all working - just been too busy to document & tidy up all the code properly, but the work in progress is here: 

https://github.com/dezfutak/thediybattery

nathan-gs, 2023-03-25:
Would swapping the S3-WIFI-ST stick with a Solis DLS-WIFI work for a RHI-4.6K-48ES-5G inverter?

Pho3niX90, 2023-11-05:
> The username is admin and the password is your wifi password - however, port 8899 is not open on the S3-WIFI-ST stick so none of these Solarman integration libraries will work. I've not yet found [...]

If the same as S4-*, then you can try port 502

honikos, 2024-07-16:
> Would swapping the S3-WIFI-ST stick with a Solis DLS-WIFI work for a RHI-4.6K-48ES-5G inverter?

Should work, works for me with Solis RHI-3P10K-HVES-5G, check the SN mentioned in readme