StephanJoubert/home_assistant_solarman #66: Avoid reset of yesterdays daily production in diagram

WallyMai, 2022-06-26:
[image]
How can I avoid the reset of the daily production at the beginning of the diagram?

pfevre, 2022-07-20:
I got the same issue on a SA3ES230 3kw mono inverter
[image]
I'm using daily production, cause total production seems not working

freirejc, 2022-07-23:
I am facing the same issue... every day in the morning when the micro inversors turn on, they make an negative reading with all produced amount from last day. I am using the sensors [sensor.deye_1_daily_production] and [sensor.deye_2_daily_production]
[image]

Mahai17, 2022-07-29:
Hi, I have the same problem with a SUN600G3 inverter. I think I found the root cause. Every night the inverter powers down (to standby?). When it starts up in the morning, the first value for total_production is "0". I think that messes up the calculations in the energy dashboard.
[image]

pfevre, 2022-07-29:
You're right Martin, I came to the same conclusion. It would be good to
handle this power down... Means connection lost to the inverter to a reset
of the daily production counter. I've seen such features in HA counters
classes. But I haven't check how to implement. Let's do it;-)

freirejc, 2022-07-29:
Yes, this is the root case.

I've found a way to workaround this issue.
I created a manual Utility Meter inside configuration.yaml, the Utility Meters I created in the Helpers didnt solve the problem.

See below the code I put, the trick for me is the line "Offset"

```
utility_meter:
  energy_deye1_daily:
    source: sensor.deye_1_daily_production
    cycle: daily
    offset:
      days: 1

  energy_deye2_daily:
    source: sensor.deye_2_daily_production
    cycle: daily
    offset:
      days: 1
```

After that I never had this negative counting in the dashboards.

[image]

[image]

Mahai17, 2022-07-29:
Are you sure that daily_production is the problem? I think, total_production is the problem, since total_production should be the input for the energy dashboard, or not? We have to prevent total_production to decrease (to "0").

freirejc, 2022-07-29:
> Are you sure that daily_production is the problem? I think, total_production is the problem, since total_production should be the input for the energy dashboard, or not? We have to prevent total_production to decrease (to "0").

This is the way I've found to have my dashboard working properly. I've tried many things.

WallyMai, 2022-07-31:
> Yes, this is the root case.
>
> I've found a way to workaround this issue. I created a manual Utility Meter inside configuration.yaml, the Utility Meters I created in the Helpers didnt solve the [...]

Works perfect for me. Thx.

leonardolisboa, 2022-08-22:
Hi, doing the above worked fine for me, however in some specific occasions there is still error in the reading, generating a huge spike (i.e: electricity provider power outages middle of the day, causing the inverter to become offline and start reporting zero in enegy). When the electricity comes back and the inverter come online again and start reporting, then the spike happens.

Is there any way to correct that at the reading, maybe making the sensor unavailable instead of reporting zero in Homeassistant?

Thanks.

Geliras, 2022-10-14:
I tried that work-around but i cant select that sensor on energy dashboard. What did i do wrong?

leonardolisboa, 2022-10-14:
> I tried that work-around but i cant select that sensor on energy dashboard. What did i do wrong?

Ensure you add in that format (including the offset) - that was the only format it appeared for me in the dashboard to be added.

energy_deye2_daily:
    source: sensor.deye_2_daily_production
    cycle: daily
    offset:
      days: 1

Geliras, 2022-10-14:
> > I tried that work-around but i cant select that sensor on energy dashboard. What did i do wrong?
> 
> Ensure you add in the format mentioned above (including the offset) - that was the only format it appeared for me in the dashboard to be added.

`utility_meter:
  energy_solar_daily:
    source: sensor.solarman_daily_production
    cycle: daily
    offset:
      days: 1`

[image]

I can't see my mistake

Floppe, 2022-10-21:
My installation had this issue also, so I will try the daily sensor now over the weekend. Hopefullt it is so easy.

However, while reading the docs. Shouldn't the state_class total_increasing never ever go backwards in HA? Or am I missing something. Perhaps it gets some bogus data which is not handled correctly on HA side that makes this happen?

paulocanedo, 2023-03-03:
same issue here

Moep90, 2023-04-28:
@Geliras @Floppe @paulocanedo Try to set it via UI.
My configured code inside the configuration file doesn't work either.