StephanJoubert/home_assistant_solarman #73: Device Serial Number is alphanumeric

NickSutton, 2022-07-13:
Hey,
Thanks for the work on this. I've got the entities created but all 'unknown' and connection status is disconnected.

The serial number for my data logger (inverter is SOLIS RHI 3-6 48ES 5G) is alphanumeric and so can't be entered in to the UI setup on the integrations page. Is this likely to be the issue preventing me from getting readings?

I can hit the local IP no issue and access the webpage, so not sure what's stopping HA from doing the same

NickSutton, 2022-07-13:
FWIW, adding the integration in YAML with the actual data logger S/N didn't work either

jmccrohan, 2022-07-15:
Looks to be a duplicate of #63

NickSutton, 2022-07-17:
Thanks, that's the type I've got. At a code level, what's the difference that stops them working with your integration?

closed 2022-07-17 by NickSutton