StephanJoubert/home_assistant_solarman #77: Solis data logger interval time and calculations

q-bhatti, 2022-07-23:
Hi.

I'm maybe being a bit stupid here but I need to know how to do the following:

1. There is no string power - I get current and voltage..how to make a sensor multiplying these in ha?
2. How do I set the refresh time to 1sec? Using visual code server I get to custom components and I can see a folder called solarman and another called inverter definitions...where do I put the sensor: but and scan_interval

gedger, 2022-07-23:
Here are my template variables to do just that. Change the sensor names to match yours.
```
- trigger:
  - platform: state
    entity_id: sensor.solis_status_lastupdate
  sensor:
  - name: PV1 DC Power
    unit_of_measurement: kW
    device_class: power
    state: >-
      {% set v = states('sensor.solis_pv1_voltage') %}
      {% set a = states('sensor.solis_pv1_current') %}
      {% if is_number(v) and is_number(a) %}
      {{ (v|float * a|float / 1000) | round(3) }}
      {% endif %}

- trigger:
  - platform: state
    entity_id: sensor.solis_status_lastupdate
  sensor:
  - name: PV2 DC Power
    unit_of_measurement: kW
    device_class: power
    state: >-
      {% set v = states('sensor.solis_pv2_voltage') %}
      {% set a = states('sensor.solis_pv2_current') %}
      {% if is_number(v) and is_number(a) %}
      {{ (v|float * a|float / 1000) | round(3) }}
      {% endif %}
```

gedger, 2022-07-23:
The minimum polling is 30 seconds, not sure why though. Can change  the code if you want to try something shorter, look in solarman.py at the line but you're on you own if it goes horribly wrong....

`@Throttle (MIN_TIME_BETWEEN_UPDATES)`

mx4k, 2022-09-07:
> - trigger:
>   - platform: state
>     entity_id: sensor.solis_status_lastupdate
>   sensor:
>   - name: PV1 DC Power
>     unit_of_measurement: kW
>     device_class: power
>     state: >-
>       {% set v = [...]

Sorry for the dumb question, but where do I put this?

gedger, 2022-09-09:
it's called a template sensor and goes into your configuration.yaml file or a file that included into it.

See. https://www.home-assistant.io/integrations/template