StephanJoubert/home_assistant_solarman #80: Data Connection Fails for Sofar HYD 6000-ES

changeableface, 2022-08-02:
I have installed and followed instructions, but unfortunately, upon choosing Sofar configuration data connection fails. I have poked around a bit and failed to make much headway. 

Some information to help debug; 
- Inverter Sofar HYD 6000-ES, firmware G330
- Wifi Stick Logger, firmware LSW3_15_FFFF_1.0.65
- Using the latest Home Assistant running via docker-compose, 
- Confirmed the container can ping the logger

Appreciate this might not be an issue you're motivated to fix, but I would obviously be super keen to see this working and happy to work with you on a solution. 

Please let me know If any further information would help diagnose this issue.

igieon, 2022-08-07:
Can you try as cusotm config file which i provide in pull request https://github.com/StephanJoubert/home_assistant_solarman/pull/82

changeableface, 2022-08-08:
Hi! 

Thanks for the pull request, unfortunately, it doesn't appear to work. 

[image]

Are there any log files I can provide to give better feedback?

Thanks again, 
- Matt

igieon, 2022-08-08:
I just look into this project https://github.com/AndyWhittaker/HYD6000/blob/master/sofar.py.

ANd based on this can you test in custom config:
```
requests:
  - start: 0x0210
    end: 0x0215
    mb_functioncode: 0x03
parameters:
  - group: Battery
      items:
        - name: "Battery Temperature"
          class: "temperature"
          state_class: "measurement"
          uom: "°C"
          scale: 1
          rule: 2
          registers: [ 0x0211 ]
          icon: 'mdi:battery'
        - name: "Grid_IO_Pwr"
          class: "power"
          state_class: "measurement"
          uom: "W"
          scale: 10
          rule: 1
          registers: [ 0x0212 ]
          icon: 'mdi:solar-power'
        - name: "House_Consumption_Pwr"
          class: "power"
          state_class: "measurement"
          uom: "W"
          scale: 10
          rule: 1
          registers: [ 0x0213 ]
          icon: 'mdi:solar-power'  
        - name: "Internal_IO_Pwr"
          class: "power"
          state_class: "measurement"
          uom: "W"
          scale: 10
          rule: 1
          registers: [ 0x0214 ]
          icon: 'mdi:solar-power'
        - name: "PV_Generation_Pwr"
          class: "power"
          state_class: "measurement"
          uom: "W"
          scale: 10
          rule: 1
          registers: [ 0x0215 ]
          icon: 'mdi:solar-power'                
```
If this works you can based on that python file add aditional info

changeableface, 2022-08-09:
Hi! 

Tried this last night and unfortunately no joy. :(

`Logger: homeassistant.components.sensor
Source: custom_components/solarman/solarman.py:34
Integration: Sensor (documentation, issues)
First occurred: 8 August 2022 at 17:42:50 (1 occurrences)
Last logged: 8 August 2022 at 17:42:50

Error while setting up solarman platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 250, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/solarman/sensor.py", line 76, in async_setup_entry
    _do_setup_platform(hass, entry.options, async_add_entities)
  File "/config/custom_components/solarman/sensor.py", line 51, in _do_setup_platform
    inverter = Inverter(path, inverter_sn, inverter_host, inverter_port, inverter_mb_slaveid, lookup_file)
  File "/config/custom_components/solarman/solarman.py", line 34, in __init__
    self.parameter_definition = yaml.full_load(f)
  File "/usr/local/lib/python3.10/site-packages/yaml/__init__.py", line 105, in full_load
    return load(stream, FullLoader)
  File "/usr/local/lib/python3.10/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
  File "/usr/local/lib/python3.10/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 82, in compose_node
    node = self.compose_sequence_node(anchor)
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 111, in compose_sequence_node
    node.value.append(self.compose_node(node, index))
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python3.10/site-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/local/lib/python3.10/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python3.10/site-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/local/lib/python3.10/site-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/local/lib/python3.10/site-packages/yaml/scanner.py", line 223, in fetch_more_tokens
    return self.fetch_value()
  File "/usr/local/lib/python3.10/site-packages/yaml/scanner.py", line 577, in fetch_value
    raise ScannerError(None, None,
yaml.scanner.ScannerError: mapping values are not allowed here
  in "/config/custom_components/solarman/inverter_definitions/sofar_g3hyd.yaml", line 7, column 12
`

igieon, 2022-08-09:
I make mistake when i generate yaml try this one:
[custom_parameters.yaml.log](https://github.com/StephanJoubert/home_assistant_solarman/files/9288409/custom_parameters.yaml.log)
I added extension log to able upload

changeableface, 2022-08-09:
Hi, 

Needed to wait till lunchtime to get back to this. Thanks for the updated file, I've dropped in place and alas still no joy. 

A get a log error detailed below:
[image]

[image]

FYI - I've verified that the docker container can see the logger by dropping into the container and pinging the IP

changeableface, 2022-08-09:
Is there any other information I could provide to better brief you on the issue, thanks for your time to date, I do appreciate it.

Matthiasfranck, 2022-08-09:
Similar issue for the `Sofar HYD 5000-ES`:

[image]

Matthiasfranck, 2022-08-09:
Seeing this error as well:

```
Traceback (most recent call last):
File "/config/custom_components/solarman/__init__.py", line 35, in update_listener
hass.data[DOMAIN][entry.entry_id].config(entry)
KeyError: 'solarman'
```

igieon, 2022-08-09:
I usually test before i create info:
I use this project:
https://github.com/jmccrohan/pysolarmanv5

install via pip  in venv if you have local python instalation then run python3 console:
```
from pysolarmanv5.pysolarmanv5 import PySolarmanV5
modbus = PySolarmanV5("add ip - from browser hostname ", 2356937823, port=8899, mb_slave_id=1, verbose=1)
#try this 
modbus.read_holding_registers(0x0210,4)
#if you get some values just compare based on https://github.com/AndyWhittaker/HYD6000/blob/master/sofar.py.
# if they make sense
```
if this works then you need look into yaml config if not then you need find some spec about registry for you inverter

changeableface, 2022-08-09:
Hi David, 

I've never done well with Python. Throwing a syntax error. Unless you spot something obvious I think we might have run aground here and have to give up - I appreciate how time-consuming and frustrating it can be trying to help the helpless, thanks for your time :)

igieon, 2022-08-09:
> ```
> >>> modbus.read_holding_registers(0x0210,4)
> DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 00 00 5f 04 7c 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 10 00 04 44 74 ab 15
> DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 10 00 10 15 00 ef 5f 04 7c 8c 02 01 ce 1e 00 00 b8 1c 00 00 26 5f f2 62 05 00 30 15 a5 01 00 10 47 00 f0 5f 04 7c 8c 00 b3 15 a5 10 00 10 15 00 f1 5f 04 7c 8c 02 01 d0 1e 00 00 ba 1c 00 00 26 5f f2 62 05 00 36 15
> DEBUG:pysolarmanv5.pysolarmanv5:frame_len does not match payload_len.
> Traceback (most recent call last):
>   File "<stdin>", line 1, in <module>
>   File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 365, in read_holding_registers
>     modbus_values = self._get_modbus_response(mb_request_frame)
>   File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 261, in _get_modbus_response
>     mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
>   File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 249, in _send_receive_modbus_frame
>     mb_response_frame = self._v5_frame_decoder(v5_response_frame)
>   File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 206, in _v5_frame_decoder
>     raise V5FrameError("V5 frame contains invalid V5 checksum")
> pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame contains invalid V5 checksum
> ```

Hi i found another documents  if you can try several time:
modbus.read_holding_registers(0x020C,4)

if result value is around 5000 its frequency of the grid 50hz if not then i found wrong documentaiton ;-/

Matthiasfranck, 2022-08-09:
> Hi i found another documents if you can try several time:
> modbus.read_holding_registers(0x020C,4)

```
>>> modbus.read_holding_registers(0x020C,4)
DEBUG:pysolarmanv5.pysolarmanv5:SENT: a5 17 00 10 45 00 00 5f 04 7c 8c 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 03 02 0c 00 04 85 b2 26 15
DEBUG:pysolarmanv5.pysolarmanv5:RECD: a5 10 00 10 15 00 f2 5f 04 7c 8c 02 01 d2 1e 00 00 bc 1c 00 00 26 5f f2 62 05 00 3b 15 a5 10 00 10 15 00 f3 5f 04 7c 8c 02 01 d4 1e 00 00 be 1c 00 00 26 5f f2 62 05 00 40 15 a5 10 00 10 15 00 f4 5f 04 7c 8c 02 01 d6 1e 00 00 c1 1c 00 00 26 5f f2 62 05 00 46 15 a5 10 00 10 15 00 f5 5f 04 7c 8c 02 01 d9 1e 00 00 c3 1c 00 00 26 5f f2 62 05 00 4c 15 a5 10 00 10 15 00 f6 5f 04 7c 8c 02 01 db 1e 00 00 c5 1c 00 00 26 5f f2 62 05 00 51 15 a5 10 00 10 15 00 f7 5f 04 7c 8c 02 01 e0 1e 00 00 ca 1c 00 00 26 5f f2 62 05 00 5c 15 a5 10 00 10 15 00 f8 5f 04 7c 8c 02 01 e2 1e 00 00 cc 1c 00 00 26 5f f2 62 05 00 61 15 a5 10 00 10 15 00 f9 5f 04 7c 8c 02 01 e4 1e 00 00 ce 1c 00 00 26 5f f2 62 05 00 66 15 a5 10 00 10 15 00 fa 5f 04 7c 8c 02 01 e6 1e 00 00 d0 1c 00 00 26 5f f2 62 05 00 6b 15 a5 10 00 10 15 00 fb 5f 04 7c 8c 02 01 e8 1e 00 00 d2 1c 00 00 26 5f f2 62 05 00 70 15 a5 10 00 10 15 00 fc 5f 04 7c 8c 02 01 ea 1e 00 00 d5 1c 00 00 26 5f f2 62 05 00 76 15 a5 10 00 10 15 00 fd 5f 04 7c 8c 02 01 ed 1e 00 00 d7 1c 00 00 26 5f f2 62 05 00 7c 15 a5 10 00 10 15 00 fe 5f 04 7c 8c 02 01 ef 1e 00 00 d9 1c 00 00 26 5f f2 62 05 00 81 15 a5 10 00 10 15 00 ff 5f 04 7c 8c 02 01 f1 1e 00 00 db 1c 00 00 26 5f f2 62 05 00 86 15 a5 10 00 10 15 00 00 5f 04 7c 8c 02 01 f3 1e 00 00 dd 1c 00 00 26 5f f2 62 05 00 8b 15 a5 10 00 10 15 00 01 5f 04 7c 8c 02 01 f5 1e 00 00 df 1c 00 00 26 5f f2 62 05 00 90 15 a5 10 00 10 15 00 02 5f 04 7c 8c 02 01 f7 1e 00 00 e1 1c 00 00 26 5f f2 62 05 00 95 15 a5 10 00 10 15 00 03 5f 04 7c 8c 02 01 f9 1e 00 00 e4 1c 00 00 26 5f f2 62 05 00 9b 15 a5 10 00 10 15 00 04 5f 04 7c 8c 02 01 fc 1e 00 00 e6 1c 00 00 26 5f f2 62 05 00 a1 15 a5 10 00 10 15 00 05 5f 04 7c 8c 02 01 fe 1e 00 00 e8 1c 00 00 26 5f f2 62 05 00 a6 15 a5 10 00 10 15 00 06 5f 04 7c 8c 02 01 00 1f 00 00 ea 1c 00 00 26 5f f2 62 05 00 ac 15 a5 10 00 10 15 00 07 5f 04 7c 8c 02 01 02 1f 00 00 ec 1c 00 00 26 5f f2 62 05 00 b1 15 a5 10 00 10 15 00 08 5f 04 7c 8c 02 01 04 1f 00 00 ee 1c 00 00 26 5f f2 62 05 00 b6 15 a5 10 00 10 15 00 09 5f 04 7c 8c 02 01 06 1f 00 00 f0 1c 00 00 26 5f f2 62 05 00 bb 15 a5 10 00 10 15 00 0a 5f 04 7c 8c 02 01 08 1f 00 00 f2 1c 00 00 26 5f f2 62 05 00 c0 15 a5 10 00 10 15 00 0b 5f 04 7c 8c 02 01 0a 1f 00 00 f5 1c 00 00 26 5f f2 62 05 00 c6 15 a5 10 00 10 15 00 0c 5f 04 7c 8c 02 01 0d 1f 00 00 f7 1c 00 00 26 5f f2 62 05 00 cc 15 a5 10 00 10 15 00 0d 5f 04 7c 8c 02 01 0f 1f 00 00 f9 1c 00 00 26 5f f2 62 05 00 d1 15 a5 10 00 10 15 00 0e 5f 04 7c 8c 02 01 11 1f 00 00 fb 1c 00 00 26 5f f2 62 05 00 d6 15 a5 10 00 10 15 00 0f 5f 04 7c 8c 02 01 13 1f 00 00 fd 1c 00 00 26 5f f2 62 05 00 db 15 a5 10 00 10 15 00 10 5f 04 7c 8c 02 01 15 1f 00 00 00 1d 00 00 26 5f f2 62 05 00 e2 15 a5 10 00 10 15 00 11 5f 04 7c 8c 02 01 18 1f 00 00 02 1d 00 00 26 5f f2 62 05 00 e8 15 a5 10 00 10 15 00 12 5f 04 7c 8c 02 01 1a 1f 00 00 04 1d 00 00 26 5f f2 62 05 00 ed 15 a5 10 00 10 15 00 13 5f 04 7c 8c 02 01 1c 1f 00 00 06 1d 00 00 26 5f f2 62 05 00 f2 15 a5 10 00 10 15 00 14 5f 04 7c 8c 02 01 1e 1f 00 00 08 1d 00 00 26 5f f2 62 05 00 f7 15 a5 10 00 10 15 00 15 5f 04
DEBUG:pysolarmanv5.pysolarmanv5:frame_len does not match payload_len.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 365, in read_holding_registers
    modbus_values = self._get_modbus_response(mb_request_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 261, in _get_modbus_response
    mb_response_frame = self._send_receive_modbus_frame(mb_request_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 249, in _send_receive_modbus_frame
    mb_response_frame = self._v5_frame_decoder(v5_response_frame)
  File "/home/matthias/.local/lib/python3.8/site-packages/pysolarmanv5/pysolarmanv5.py", line 204, in _v5_frame_decoder
    raise V5FrameError("V5 frame contains invalid start or end values")
pysolarmanv5.pysolarmanv5.V5FrameError: V5 frame contains invalid start or end values
```

sjoerdos92, 2022-08-14:
I managed to fix @igieon's file for my use case, see my comment here: https://github.com/StephanJoubert/home_assistant_solarman/pull/82#pullrequestreview-1072112654

changeableface, 2022-08-24:
If you can, I'd be keen to see your integration. 

I have API details to query the SolarMan website, while useful, information from this source is not real-time, instead around 6 to 7 minutes out of date with fewer data points. 

Thanks @Matthiasfranck !

Matthiasfranck, 2022-08-24:
Indeed, using the API was something I wanted to avoid because of this massive delay. Give me some time to make the integration ready to be shared.

The connection to the serial interface is 100% based on this: https://github.com/AndyWhittaker/HYD6000
With this cable: https://www.amazon.de/-/nl/gp/product/B089LWCR4G/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1

changeableface, 2022-08-24:
Thanks champ, I'll get one ordered in the meantime.

changeableface, 2022-08-31:
Hi @Matthiasfranck 

I'm up and running now, thanks for pointing me in the right direction. 

For future people who find this comment thread; 

I purchased a USB RS485 dongle, different to the link but just a generic one in the UK that would get here in a few days, and used the [EMONPI image](https://github.com/openenergymonitor/emonpi/wiki/emonSD-pre-built-SD-card-Download-&-Change-Log#emonsd-24jul20-beta) with a Pi3 and the [code here](https://github.com/AndyWhittaker/HYD6000/blob/master/sofar.py) (with an update to line 116 make it work with Python3) 

`from urllib.parse import urlparse`

Surprisingly, this got the data into EmonPi without much more faff. 

A quick 3-line bash script to get this to run every 5 seconds from reboot (via cron) and then from there I updated home assistants configurations.ymal to include; 

`# Solar PV
sensor:
  platform: emoncms
  api_key: XXXXXXXXX
  url: http://192.168.10.106
  id: 1`

The Information pulled in doesn't have unique entity ID's so there's some work I need to do to make that happen, but that's it, the data is where I want it. A complete pig to get it there, but obviously great fun too. 

Thanks for pointing me in the right direction. I will close this comment as wontfix.

closed 2022-08-31 by changeableface

changeableface, 2022-09-19:
Hi! 

Yeah I do and yeah it's working. 

Couple of pictures for you. 
[image]
[image]
[image]
[image]

Matthiasfranck, 2022-09-20:
@changeableface Thanks. The technician came here yesterday to install my batteries. The system is going into alarm and the batteries won't work. The technician is blaming me of braking the inverter by using the 485 port.

changeableface, 2022-09-20:
While I'm no technician, that seems unlikely given the circumstance. 

Good luck - let me know if there's anything I can provide that might prove useful on your quest!