StephanJoubert/home_assistant_solarman #86: Deye Micro Inverter SUN2000G3

alessandroo23, 2022-08-14:
Has anyone managed to integrate Deye SUN2000G3 microinverters?

leonardolisboa, 2022-08-22:
Here https://github.com/StephanJoubert/home_assistant_solarman/issues/46

kababoom, 2022-10-27:
I have the 1600 version and it's working fine with below config, of course you need to add the deye_4mppt.yaml to the inverters definitions..

There's just one thing that I cannot change and that's the interval, it stays on every 5 minutes whatever setting I change...
```
  - platform: solarman
    name: DEYE
    inverter_host: 192.168.10.90
    inverter_port: 8899
    inverter_serial: 4130xxxxxx
    inverter_mb_slaveid: 1
    scan_interval: 30
    lookup_file: deye_4mppt.yaml

  - platform: template
    sensors:
      deye_pv1_power:
        unit_of_measurement: W
        device_class: power
        value_template: "{{ states('sensor.deye_pv1_voltage') | float * states('sensor.deye_pv1_current') | float }}"

      deye_pv2_power:
        unit_of_measurement: W
        device_class: power
        value_template: "{{ states('sensor.deye_pv2_voltage') | float * states('sensor.deye_pv2_current') | float }}"

      deye_pv3_power:
        unit_of_measurement: W
        device_class: power
        value_template: "{{ states('sensor.deye_pv3_voltage') | float * states('sensor.deye_pv3_current') | float }}"

      deye_pv4_power:
        unit_of_measurement: W
        device_class: power
        value_template: "{{ states('sensor.deye_pv4_voltage') | float * states('sensor.deye_pv4_current') | float }}"

```

danilodene, 2022-11-15:
@kababoom  same  here

DHLF, 2023-04-23:
Deye microinverter polling time is set to 5 minutes(default).
If you want to change(shorten) this you MUST contact Deye(registration, serial number, etc. needed) and left your device connected to the internet(they will change it from afar..)
AFAIK the shortest time would be 1 minute.

Rumours came up about WRITING to the microinverter with post commands may be possible but be aware:
wrong registers-wrong values may **cause unexpected harm**- to YOU, your family or neighbours lifes!

Better stay at those time limits or contact deye.

DHLF, 2023-04-23:
@kababoom 
Your solution looks great(adding POWER as I*U for the 4 PV) but..  I dont know where to put those stuff..
Just adding the lines into the deye_4mppt.yaml or into the /root/config/config.yaml?

kababoom, 2023-04-29:
> @kababoom 
> 
> Your solution looks great(adding POWER as I*U for the 4 PV) but..  I dont know where to put those stuff..
> 
> Just adding the lines into the deye_4mppt.yaml or into the /root/config/config.yaml?
> 
> 
> 
> As my /root/config/config.yaml does NOT contains stuff about the solarman integration..

This goes in configuration for example..

[image]