StephanJoubert/home_assistant_solarman #87: For info : Port is not always 8899

gdwa, 2022-08-14:
I have a Solis Hybrid Firmware version ME-121001-V1.0.6(202206141010)

For those having Unknown Status, give a try at looking at the interface of your datalogger.

In the tab Work Mode :
- Port of the device 30003
I had to change 8899 default value to that one and everything is now working fine.

Rgds

jmccrohan, 2022-08-15:
@gdwa Is this a DLS-L (ethernet logging stick)?

Are you able to post screenshots of the various webUI screens that you needed to change to get it to work?

gdwa, 2022-08-16:
@jmccrohan 
It’s a Ethernet logger indeed. 

I made no change at all to the logger itself, I only adapted the port in the Solarman config flow to match the one of the logger web UI

GillesC, 2022-11-10:
> I have a Solis Hybrid Firmware version ME-121001-V1.0.6(202206141010)
>
> For those having Unknown Status, give a try at looking at the interface of your datalogger.
>
> In the tab Work Mode :
>
> * Port of [...]

Where did you find the used port? The WebUI does not to show which port is used.

gdwa, 2022-11-10:
[image]

Please note that I'm working in local mode and no longer local + remote. This improved incredibly the stability of data returned to HA. I lost the cloud (iOS app) which was crappy anyway.

GillesC, 2022-11-10:
Thank you, I do not have a work mode option. I was also looking into ways to skip the cloud and do everything locally.