StephanJoubert/home_assistant_solarman #88: Deye Sun SUN-12K-SG04LP3

wendtc, 2022-08-18:
Hello,
Great project you have here. I have a Deye 12K hybrid inverter. Unfortunately, your YAML file for Deye did not display anything.
I got the Modbus protocol from Deye and adjusted the first things.
However, I have not been able to find battery status, inverter status, etc.
Maybe with your experience you can find out these things.
Christia
[deye12KEU.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/9368217/deye12KEU.yaml.txt)
[MODBUS RTUDEYEübersetzt.pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/9368219/MODBUS.RTUDEYEubersetzt.pdf)
n

PV-Joe, 2022-09-05:
I would highly appreciate if anyone would find out the missing values as this configuration fits to my Deye SUN-8K-SG04LP3-EU as well.

Brilliant work Wendttc. 👍

PV-Joe, 2022-09-05:
[image]

This register doesn’t seem to work. Has anyone got a clue what value has to go in there to make the sensor work?

wendtc, 2022-09-11:
[deye12K.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/9543319/deye12K.yaml.txt)
New Register on Deye 12K

Cerak321, 2022-09-16:
Correct register for: 
Daily production: 0x0211
Battery power: 0x024E

All these values can be found in [this document](https://github.com/maxx-ukoo/deye-modbus2mqtt/blob/master/docs/Modbus%20amit%20posun.docx) you just have to convert decimal values from the left column to hexadecimal and don't forget to add these registers to request if they are not there.

PV-Joe, 2022-09-17:
Hey Cerak,

great work! Thx for posting.
The daily production works fine ✅
The battery power is still „unknown“ ❌

Here is my configuration for battery power:
[image]

Mia there another value I could try?

Joe

PV-Joe, 2022-09-19:
Hey Cerak,
I just realized that:

Daily production: 0x0211
… is not production.

It is:
Daily energy charged

Are you able to correct this?

ondrejbartas, 2022-09-29:
@wendtc I love you ❤️  

I spend past 4 days looking for way to get data from LSW3 logger 23XXX in Deye SUN-10K EU. And most of data is correct !!!! You are life saver.

ondrejbartas, 2022-09-30:
Total Energy Sold update:

```
    - name: "Total Energy Sold"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 3
      registers: [0x020C,0x020D]
      icon: 'mdi:transmission-tower-import'
```

changed registers `[0x020C,0x020D]`

PV-Joe, 2022-09-30:
> Total Energy Sold update:
> 
> ```
>     - name: "Total Energy Sold"
>       class: "energy"
>       state_class: "total_increasing"
>       uom: "kWh"
>       scale: 0.1
>       rule: 3
>       registers: [0x020C,0x020D]
>       icon: 'mdi:transmission-tower-import'
> ```
> 
> changed registers `[0x020C,0x020D]`

Excellent work. Just tested it and it works for my DEYE SUN-8K-SG04LP3-EU

Thank you so much!
Keep going and I would be happy if you could post any other corrections here in future 👍

PV-Joe, 2022-10-01:
Hi again,

my Stick-Logger was changed to a newer model today because upgrading the firmware per remote wasn't working.

Now I have got a Stick-Logger with the following firmware:
**LSW3_15_FFFF_1.0.91R**
Serialnumber beninning with: **27xxxxxxx**

The result is that all the values in Home Assistant are not working anymore even though I changed the new IP and Serialnumber in the configuration. The Port ist still 8899.

Any ideas what I have to do to make the new LSW-3 Stick-Logger work again?

ondrejbartas, 2022-10-01:
I found ot problem with Battery Power and Current:

correct settings are:

```

    - name: "Battery Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 2
      registers: [0x024E]
      icon: 'mdi:battery'

    - name: "Battery Current"
      class: "current"
      state_class: "measurement"
      uom: "A"
      scale: 0.01
      rule: 2
      registers: [0x024F]
      icon: 'mdi:battery'

```

But you also need to change in beggining of file requests:
```
  - start: 0x024A
    end: 0x024D
```

to 

```
  - start: 0x024A
    end: 0x024F
```
[image]

PV-Joe, 2022-10-01:
> Hi again,
>
> my Stick-Logger was changed to a newer model today because upgrading the firmware per remote wasn't working.
>
> Now I have got a Stick-Logger with the following firmware: [...]

Got it working in the meantime 😄
The problem was, that after changing the serialnumber all sensors were duplicated. I deleted the old ones and deleted the xxx_2 for the new sensors and voila. It works again. 

Thank you for helping myself 😂

PV-Joe, 2022-10-01:
> I found ot problem with Battery Power and Current:
>
> correct settings are:
>
> ```
>
>     - name: "Battery Power"
>       class: "power"
>       state_class: "measurement"
>       uom: "W"
>       scale: 1 [...]

I just changed the registers and the output is:
[image]

It looks strage to me 🤔

ondrejbartas, 2022-10-01:
If power is negative tahn you are most probably chargin battery

PV-Joe, 2022-10-02:
> If power is negative tahn you are most probably chargin battery

You are right. :-)
Thank you for your great work!

PV-Joe, 2022-10-05:
> Another good metric:
> 
> ```
>     - name: "Grid side total power"
>       class: "power"
>       state_class: "measurement"
>       uom: "W"
>       scale: 1
>       rule: 2
>       registers: [0x0271]
>       icon: 'mdi:transmission-tower'
> ```
> 
> and also in requests:
> 
> ```
> - start: 0x0276  
>     end: 0x027C
>     mb_functioncode: 0x03  
> ```
> 
> change to
> 
> ```
>   - start: 0x0271
>     end: 0x027C
>     mb_functioncode: 0x03
> ```
> 
> this will give you power on grid (positive is buy, negative is sell) It should be same number which is shown on display of inverter (on grid side)

Hi,

another excellent work. Tested it on my Deye  8K Hybrid and it works perfectly. I helped myself out by summing up the three LT1 to LT3 values in configuration.yaml creating a sensor with the following code until now:

`# Sensor für Solarman CT1-3 Sum
      solar_external_ct1_3:
        friendly_name: "Solarman CT1-3 Power"
        unit_of_measurement: 'W'
        value_template: "{{ states('sensor.solarman_external_ct_l1_power')|int(0) + states('sensor.solarman_external_ct_l2_power')|int(0) + states('sensor.solarman_external_ct_l3_power')|int(0) }}"
`
This is obsolete now.

Thank you 😄

Magno06, 2022-11-20:
@ondrejbartas Thanks for sharing parts of the setup for the 10K, I have one too. It works but to have the total PV production, what should I write? Now returns "unavailable". Thanks again

Magno06, 2022-11-20:
@ondrejbartas Thanks, everything works. I also have a DEYE 10K but I can't see the daily production, it still returns "unknow" and I don't know how to edit the .yaml file. can you help me? Have a good evening

PV-Joe, 2022-11-20:
> @ondrejbartas Thanks, everything works. I also have a DEYE 10K but I can't see the daily production, it still returns "unknow" and I don't know how to edit the .yaml file. can you help me? Have a good evening

Try this in your “deye_hybrid.yaml”:

    - name: "Daily Production"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 1
      registers: [0x0211]
      icon: 'mdi:solar-power'

Use the Addon “File Editor” to edit the yaml.

tomasvitha, 2022-11-23:
Great work guys, thank you! Works perfectly with a new DEYE 10K-SG04LP3-EU hybrid.

Magno06, 2022-11-24:
> Try this in your “deye_hybrid.yaml”:
>
> ```
> - name: "Daily Production"
>   class: "energy"
>   state_class: "total_increasing"
>   uom: "kWh"
>   scale: 0.1
>   rule: 1
>   registers: [0x0211]
>   icon: [...]

Thanks for your invaluable help! Another question ... Is there a way to have a sensor that indicates the daily consumption of my house? I see it in the energy distribution graph but I don't have a sensor that I can use. can you help me? Thanks again
[image]

PV-Joe, 2022-11-24:
You are nearly ready 😄
Try:
sensor.solarman_daily_load_consumption

Klagio, 2022-11-30:
Hi, I have exactly the same Inverter (actually 2, one master and one slave).   Can you please help me on how to change the component so that it works?

Klagio, 2022-11-30:
> Thanks for your invaluable help! Another question ... Is there a way to have a sensor that indicates the daily consumption of my house? I see it in the energy distribution graph but I don't have a [...]

Hi just to make sure I don't do anything wrong, these are the 5 sensors for the graph? To be put in recorder.yaml?

      - sensor.solarman_daily_production
      - sensor.solarman_daily_load_consumption
      - sensor.solarman_total_battery_charge
      - sensor.solarman_total_battery_discharge
      - sensor.solarman_daily_production

PV-Joe, 2022-12-06:
**Here is the full working deye_hybrid.yaml**

[deye_hybrid.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/10169888/deye_hybrid.yaml.txt)

[image]

Klagio, 2022-12-07:
> **Here is the full working deye_hybrid.yaml**
> 
> [deye_hybrid.yaml.txt](https://github.com/StephanJoubert/home_assistant_solarman/files/10169888/deye_hybrid.yaml.txt)
> 
> [image]

Thanks. Very cool dashboard, you mind sharing the lovelace code?

Klagio, 2022-12-07:
I have this as differenzce
```

    - name: "SmartLoad Enable Status"
      class: ""
      state_class: ""      
      uom: ""
      scale: 1
      rule: 1
      registers: [0x0085]
      isstr: true
      lookup: 
      -  key: 0
         value: "GEN Use"
      -  key: 1
         value: "SMART Load output"
      -  key: 2
         value: "Microinverter"
      icon: 'mdi:lightning-bolt-outline'
```

Klagio, 2022-12-07:
which is the correct?
```

    - name: "Total Grid Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 2
      registers: [0x0271]
      icon: 'mdi:transmission-tower'
    - name: "Total Grid Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 2
      registers: [0x0260]
      icon: 'mdi:transmission-tower'
```

Magno06, 2022-12-26:
Good morning everyone and happy holidays,
I had correctly modified the DEYE 10K inverter yaml file but I had lost the integration and therefore the code. Would someone with the same inverter (@ondrejbartas) as me be so kind as to provide me with the complete code? I would be very grateful :)

linuxdep, 2022-12-29:
Hi, thanks for add correct parameter for deye12kEU, but i have found some other issues:

[sensor.deye12k_battery_temperature] wrong i see -86°C
[sensor.deye12k_grid_connected_status] wrong status GRID, i see "unavailable" but it is connect to grid
[sensor.deye12k_time_of_use] i see "unavailable"
[sensor.deye12k_total_power] i see "unavailable"

have you a fix for me?

Derirrepeter, 2023-03-26:
> Got it working in the meantime 😄 The problem was, that after changing the serialnumber all sensors were duplicated. I deleted the old ones and deleted the xxx_2 for the new sensors and voila. It [...]

Can you give me some help i have the same Inverter and have verry poor experience with HA and Solarman. I have the HACS ans Solarman running and tried to integrate the new Deye_12K.yaml into my HA and the Solarman. this didnt worked.
Please i need help.

sdrogi, 2023-05-30:
First of all thank you for this great work....
With this thread I got my 10K-SG04LP3-EU connected to my Home Assistant. I have only one small problem left. In addition to the PV 1 and PV 2 strings, there are also three micro inverters connected to and controlled by the SG04LP3-EU. The output in the app is displayed separately. But I can't find the matching resgisters in the yaml or the modbus file. Does anyone have an idea about this?
Thanks in advance
[image]
[image]

K3iVi, 2023-07-31:
> First of all thank you for this great work.... With this thread I got my 10K-SG04LP3-EU connected to my Home Assistant. I have only one small problem left. In addition to the PV 1 and PV 2 strings, [...]

I can not find total numbers, but you can create meter.

Edit deye_sg04lp3.yaml:

requests:
  - start: 0x0295 
    end: 0x029B
    mb_functioncode: 0x03

and add sensors:

    - name: "Phase voltage of Gen port A"
      class: "voltage"
      state_class: "measurement"      
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0295]
      icon: 'mdi:home-lightning-bolt'

    - name: "Phase voltage of Gen port B"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0296]
      icon: 'mdi:home-lightning-bolt'
      
    - name: "Phase voltage of Gen port C"
      class: "voltage"
      state_class: "measurement"
      uom: "V"
      scale: 0.1
      rule: 1
      registers: [0x0297]
      icon: 'mdi:home-lightning-bolt'
      
    - name: "Phase power of Gen port A"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x0298]
      icon: 'mdi:home-lightning-bolt'
      
    - name: "Phase power of Gen port B"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x0299]
      icon: 'mdi:home-lightning-bolt'

    - name: "Phase power of Gen port C"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x029A]
      icon: 'mdi:home-lightning-bolt'
      
    - name: "Total Power of Gen port"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1
      rule: 1
      registers: [0x029B]
      icon: 'mdi:home-l1ghtning-bolt'

sdrogi, 2023-07-31:
Great ... Thank you

K3iVi, 2023-08-03:
> 

Good news everyone :) Found out registers for Microinverter generation :)
  - start: 0x0218
    end: 0x021A
    mb_functioncode: 0x03

    - name: "Microinverter power generation of today"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 3
      registers: [0x0218]
      icon: 'mdi:transmission-tower-import'

    - name: "Microinverter power generation total"
      class: "energy"
      state_class: "total_increasing"
      uom: "kWh"
      scale: 0.1
      rule: 3
      registers: [0x0219,0x021A]
      icon: 'mdi:transmission-tower-import'

Hieu137, 2023-09-21:
> I can not find total numbers, but you can create meter.
>
> Edit deye_sg04lp3.yaml:
>
> requests:
>
> * start: 0x0295
>   end: 0x029B
>   mb_functioncode: 0x03
>
> and add sensors:
>
> ```
> - name: "Phase voltage of [...]

Hi , Can you share with me Modbus documents ?

K3iVi, 2023-09-21:
This is the last one I got from Deye support. Totals on microinverters are hidden under Chinese 536,537,538.
Singles GEN port phases are from 661+

[Deye MODBUS RTU三相储能通信规约V104 (1).pdf](https://github.com/StephanJoubert/home_assistant_solarman/files/12682805/Deye.MODBUS.RTU.V104.1.pdf)

chinezbrun, 2023-10-14:
hello guys. great work on this tread issue. 

Is the original deye_sg04lp03.yaml updated with these values in curent release, or all changes needs to be done manually?

If yes, what's the meaning of the fields marked below in the yaml file?

ie:
requests:                              # what does means
  - start: 0x0003                    # what does means
    end: 0x0059                     # what does means
    mb_functioncode: 0x03   # what does means

parameters:
 - group: solar
   items: 
    - name: "PV1 Power"
      class: "power"
      state_class: "measurement"
      uom: "W"
      scale: 1                                          # what does means
      rule: 1                                            # what does means
      registers: [0x02A0]                        # obvious this is the registry in HEX but i saw also two values here....

hajokim, 2024-02-01:
> Got it working in the meantime 😄 The problem was, that after changing the serialnumber all sensors were duplicated. I deleted the old ones and deleted the xxx_2 for the new sensors and voila. It [...]

Hello, I have the same problem. But I don't understand how do you resolve this problem? I would be very grateful, if you send me the solution. Thank you.