StephanJoubert/home_assistant_solarman #93: Unstable Connection and wrong values

gintonic26, 2022-08-25:
Hello, and first of all thank you very much for you add-on to HA. 

I have a small 600W inverter with two 335W photovoltaic modules. 
The inverter is connected to Solarman Server via WLAN, which works quite okay but I can see some values. 
My Dashboard and the webpage say 4.9 kWh total production ( The inverter is up the fourth day)
[image]
[image]

I did the installation with HACS and the config through the dashboard:
[image]
With "deye_hybrid.yaml" i could get no connection at all, and somewhere i found someone who has used "sofar_lsw3.yaml".
Now I have a bit unstable connection. 
[image]

And now the funny part begins:
[image]

Official Dashboard 4,9 kWh / Integration 3,9 kWh

Does anybody have a good idea? 

Thanks and regards
Gin

awalon, 2022-09-28:
Got the kWh vs W issue fixed with my PR #105 but still same issue with "old" for first measurement per day. I think value was reset based on missing or wrong value during hours without sunlight.

awalon, 2022-09-30:
Wrong values (start/end of day) can be fixed by this [workaround](https://github.com/StephanJoubert/home_assistant_solarman/issues/66#issuecomment-1199069292) (from #66)

VaillantHassIo, 2022-11-03:
this one works for Bosswerk MI600 inverters
https://github.com/StephanJoubert/home_assistant_solarman/issues/46