StephanJoubert/home_assistant_solarman #96: Request to copy your code

AdyRock, 2022-09-05:
Hi,

Would you mind if I copy your code and convert it to Node.js for use with Homey?

And to be really cheeky, could you provide a typical request packet for a dummy sn, etc, so I can compare the output from my code to ensure I have got things right?

StephanJoubert, 2022-09-05:
Hi @AdyRock , no problem. I wanted to do Node.js as well, but too much stuff on my plate as usual. Will see if I can get hold of some sequences for you.

AdyRock, 2022-09-05:
Thank you, much appreciated.
How do you get the MODBUS register information for various inverters?
I have a Sofar HYD 3600-ES so not sure if the description files in your repo will work for me.

AdyRock, 2022-09-07:
I have had some success. My inverter has a different register layout, so I have made a new description file and I now get valid data back most of the time.
However, I find occasionally the data is crazy and it seem that the response has the wrong number of bytes. The Solarman V5 packets are perfectly formed with the correct check sum, etc, but the MODBUS payload is fragmented within it, so the first V5 data packet returns the start of the MODBUS registers and then, after the next request, I get a second V5 packet with the remaining MODUS registers. As the code doesn't verify the MODBUS CRC, the result is it decodes the values incorrectly.

Have you seen anything like this?

StephanJoubert, 2022-09-07:
I have not seen that behavior exactly, but found that when the registers has a gap, it works better if the request is split into multiple  requests, with each one requesting contiguous registers.

AdyRock, 2022-09-07:
Yep, I have done that. Maybe I will reduce the block size as my first block goes from 512 to 547, which, in theory, all contain valid data.

igieon, 2022-09-07:
@AdyRock  i think modbus protocol (registers) is same as in my inverter only that is single phase so your interval is not same as me https://github.com/StephanJoubert/home_assistant_solarman/blob/main/custom_components/solarman/inverter_definitions/sofar_g3hyd.yaml. And when i look into your range 0x512-0x547 its outside of values so probably look into this file and mask which is correct register adress. I think file is correct https://www.facebook.com/groups/2477195449252168/permalink/2895353217436387/.

AdyRock, 2022-09-07:
Ah, sorry, the register numbers I gave are in decimal (JSON can't cope with hex), so the a
<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Adrian/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Adrian/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link="#0563C1" vlink="#954F72">

  |   |  
-- | -- | --
Running state | 512 | 200
Fault message 1 | 513 | 201
Fault message 2 | 514 | 202
Fault message 3 | 515 | 203
Fault message 4 | 516 | 204
Fault message 5 | 517 | 205
Grid A Voltage | 518 | 206
Grid A Current | 519 | 207
Grid B Voltage | 520 | 208
Grid B Current | 521 | 209
Grid C Voltage | 522 | 20A
Grid C Current | 523 | 20B
Grid Frequency | 524 | 20C
Charge/Discharge power | 525 | 20D
Battery Voltage | 526 | 20E
Battery Charge/Discharge current | 527 | 20F
The residual capacity of battery | 528 | 210
Battery Temperature | 529 | 211
Feed in/out power | 530 | 212
The power of the load | 531 | 213
Input/Output power | 532 | 214
The power of generation | 533 | 215
EPS output voltage | 534 | 216
EPS output power | 535 | 217
Generation of one day | 536 | 218
The power sell to grid of today | 537 | 219
The power buy from grid of toda | 538 | 21A
Today consumption of the load | 539 | 21B
Total generation H Byte | 540 | 21C
Total generation L Byte | 541 | 21D
Total power sell to grid H Byte | 542 | 21E
Total power sell to grid L Byte | 543 | 21F
Total energy buy from grid H Byte | 544 | 220
Total energy buy from grid L Byte | 545 | 221
Total Consumption H Nyte | 546 | 222
Total Consumption L Byte | 547 | 223
Reseved | 548 | 224
Reseved | 549 | 225
Reseved | 550 | 226
Reseved | 551 | 227
Reseved | 552 | 228
Reseved | 553 | 229
The countdown time | 554 | 22A
Inverter alert message | 555 | 22B
Battery Cycle times | 556 | 22C
INV bus voltage | 557 | 22D
LLC bus voltage | 558 | 22E
Buck current | 559 | 22F
Grid R Voltage | 560 | 230
Grid R Current | 561 | 231
Grid S Voltage | 562 | 232
Grid S Current | 563 | 233
Grid T Voltage | 564 | 234
Grid T Current | 565 | 235
Generation current | 566 | 236
Battery power | 567 | 237
Inner temperature | 568 | 238
Heat sink temperature | 569 | 239
Country | 570 | 23A
Current dc component | 571 | 23B
Voltage dc component | 572 | 23C
Battery fault message 1 | 573 | 23D
Battery fault message 2 | 574 | 23E
Battery fault message 3 | 575 | 23F
Battery fault message 4 | 576 | 240
Battery fault message 5 | 577 | 241
Communication board inner message | 578 | 242
Today generation time | 579 | 243
Total generation time High-Byte | 580 | 244
Total generation time,Low-Byte | 581 | 245

</body>

</html>

AdyRock, 2022-09-07:
After splitting the ranges a bit more, I'm still getting the occasional split MODBUS packet. The data is all good, it's just the first V5 packet is missing the data from the last few registers from the MODBUS payload and the second V5 packet contains the remaining register data from the first request plus the data from the second request.