StephanJoubert/home_assistant_solarman #97: I am not getting some values (DEYE Daily Energy Bought/Sold) - sofar_lsw3

Erestes, 2022-09-08:
I am running Sofar inverter 5,5 since some moths and successfully collecting a lot of useful data, but the most important ones on which the Energy card depends on are not showing up. I don't see: DEYE Daily Energy Bought and DEYE Daily Energy Sold for instance. 
What may be the reason for this and where should I start troubleshooting?

Thanks in adv.

PV-Joe, 2022-09-08:
It is simular with the Deye 8K Hybrid inverter. I’m missing daily production/ total production and hope for someone to figure out those two Modbus RTU registers that only appear as “unavailable” in Home Assistant. Most of the other values are OK and working incl. daily energy bought and daily energy sold.

chawawa, 2023-01-18:
> It is simular with the Deye 8K Hybrid inverter. I’m missing daily production/ total production and hope for someone to figure out those two Modbus RTU registers that only appear as “unavailable” in [...]

I have the Deye 8k Hybrid. Been running for 3 months now and collecting all my data. I cant sell back to the grid so I cant show if that is working. I got mine working after setting up my energy dashboard. (if you not sure were to set it up go to Setting -> dashboards -> Energy. Add the sensors as per my picture.

[image]

I then have this data output into a card for quick reference and for more detail i use the energy dashboard
[image]

Hope this helps.

Side note: On the Deye 8K Hybrid most people have two strands coming from there roof. You can combine the two PV senors to give you a total value. add the below to your config and you will see a new senor called "Solar Generation - Total"

#Total Solar power from roof
- platform: template
  sensors:
    powermon_totalsolar:
      friendly_name: "Solar Generation - Total"
      unit_of_measurement: "watts"
      value_template: >- 
        {{ (states('sensor.solarman_pv1_power') | float + states('sensor.solarman_pv2_power') | float) }}
      availability_template: >-
        {{ states('sensor.solarman_pv1_power') not in ['unknown', 'unavailable'] and states('sensor.solarman_pv2_power') not in ['unknown', 'unavailable'] }}