StephanJoubert/home_assistant_solarman #98: Total Increasing Sensors should be made unavailable when the inverter gets disconnected to play well with Energy Dashboard.

FSciammarella, 2022-09-09:
When a sensor status is set to unknown or none, it is regarded as zero. When a total_increasing value decreases, even if by error, homeassistant treats this as a reset from 0, leading to wrong data.

FSciammarella, 2022-09-09:
A simple workaround I used was modifying the update function of SolarmanSensorText as follows
`    def update(self):
    #  Update this sensor using the data.
    #  Get the latest data and use it to update our sensor state.
    #  Retrieve the sensor data from actual interface
        self.inverter.update()
        if self.inverter.status_connection == "Disconnected":
            self.available = False
            return
        self.available = True
        val = self.inverter.get_current_val()
        if val is not None:
            if self._field_name in val:
                self.p_state = val[self._field_name]`

FSciammarella, 2022-09-09:
Also, to avoid unknown status on startup, sensor should be set to Unavailable upon initialization with 
```python
# ...
class SolarmanStatus(Entity):
    def __init__(self, inverter_name, inverter, field_name, sn):
        self._attr_available = False
# ...
```

rolandsteinmeyer, 2022-10-07:
Will try this modification. My deye 1600 shuts down when sun is down and total production briefly goes to 0 at inverter start in the mprning, which messes up energy dashboard.
[image]
[image]

rolandsteinmeyer, 2022-10-07:
Two side effects after patching sensors.py as described above:

I get log entries "Update of sensor.solarman_pv1_voltage is taking over 10 seconds". This is an unused port on my 4-Port inverter. Hence voltage is always 0. 

Sensors status_connection and status_lastUpdate are now always "unknown".

rolandsteinmeyer, 2022-10-09:
Patch does not work for me. Total Production still drops  to zero in the morning.
[image]

pfmacedo, 2022-10-10:
> Patch does not work for me. Total Production still drops to zero in the morning. 

I think the patch works for the specific issue of the inverter becoming unreachable; in the case shown in your screenshot, it's a slightly different issue, which I've noticed on my install as well (with sun2000g3-us-220 inverters): at boot up, they return 0 for most values for a few minutes (you can see this for yourself by going to the inverter UI, telling it to reboot and then check the status page as soon as it's reachable).

The fix for that is IMO to have solarman return 'unavailable' if the inverter is returning any 0 values that should never become zero, like historical production. 

As for a work-around, I've used a template to ignore 0/"unavailable"/"unknown" and that has fixed the energy dashboard for me.

rolandsteinmeyer, 2022-10-12:
I am using the workaround from #66 now. A template sensor probably would also do the trick. Also thinking about using a Riemann sum integral  sensor on AC Output Power, which would offer finer steps than 0.1 kWh but would miss any energy produced when connection from HA to inverter is broken.

cylonbrain, 2022-10-13:
> I am using the workaround from #66 now. A template sensor probably would also do the trick. Also thinking about using a Riemann sum integral sensor on AC Output Power, which would offer finer steps than 0.1 kWh but would miss any energy produced when connection from HA to inverter is broken.

I don't think you will increase your precission, because the AC Output Power updates like once every 4-5 minutes, but the actual output power can oscillate faster than that.

Geliras, 2022-10-13:
> As for a work-around, I've used a template to ignore 0/"unavailable"/"unknown" and that has fixed the energy dashboard for me.
Can you please show me how you did that?

Do you possibly have a workaround for a sensor that always shows the total_production even if the inverter is offline?

pfmacedo, 2022-10-15:
> Can you please show me how you did that?

Sure, here is one of the inverters:

```
template:
  - sensor:
    - name: "Inverter 1 energy total"
      unique_id: inverter1_energy_total
      state_class: total_increasing
      device_class: energy
      unit_of_measurement: kWh
      icon: mdi:solar-power
      state: >
          {% if states('sensor.inversor1_total_production')  in ('unavailable', 'unknown') %} 
            {{ states('sensor.inverter1_energy_total') }}
          {% else %}
            {% if states('sensor.inversor1_total_production') | float == 0 %}
              {{ states('sensor.inverter1_energy_total') }}
            {% else %}
              {{ states('sensor.inversor1_total_production') | float | multiply(1.001) | round(1) }}
            {% endif %}
          {% endif %}
```

sensor.inversor1_total_production is the solarman variable, sensor.inverter1_energy_total is the template/derived one (I have a bad habit of mixing portuguese and english on my configs )

The resulting variable is always increasing/non-zero, but it does go away if the inverter is unreachable, as can be seen in this screenshot:

[image]