StephanJoubert/home_assistant_solarman #99: option to set password

michaelruck138, 2022-09-11:
Hello,
I've tried getting this working with my 2 solis inverter data loggers. their serial numbers start with 404 and 408.
In an effort to secure things I always change any default password and everything I've read about this integration uses a username and password of admin, however that isn't my password.
I was wondering if there was anywhere in the configure where I could set the password? as at the moment no matter what setup and port number I pick all the values come back as unknown.
tia!

r3pl1x, 2023-03-06:
Same problem with Deye SUN600G3-EU-230 and Deye SUN800G3-EU-230.
Had to revert to default login credentials.

stefmat10, 2023-03-13:
Any update to enable us to set the password as I have the same issue with my Solis stick too???

enjoyitapp, 2023-03-29:
same here, please let us know howto fix this. Its bad network practice to leave default passwords.

UserTotheMoon2, 2023-06-14:
I also want to request to have the username/password changeable. Using admin/admin in deye is highly unsecure

frsantos, 2023-06-14:
Not sure what username/password you are referring to, but as far as I know, this integration uses no password at all.

osulivanDE, 2023-06-14:
> Not sure what username/password you are referring to, but as far as I know, this integration uses no password at all.

That´s the problem. As soon as you change the credentials on the inverter the integration is unable to load data from it. The Integration uses admin/admin as default credentials and it is impossible to give it other credentials. 
So you stuck on admin/admin as long as you want to use the solarman HA integration.

frsantos, 2023-06-14:
> The Integration uses admin/admin as default credentials and it is impossible to give it other credentials. So you stuck on admin/admin as long as you want to use the solarman HA integration.

The integration does not use any password at all, as you can see if you search the code for "admin":  https://github.com/search?q=repo%3AStephanJoubert%2Fhome_assistant_solarman%20admin&type=code

The "admin" password is the one that the inverter uses to show its web interface, but this project does not connect to that web interface. It uses the modbus protocol that is exposed without password.

Not sure why it stops working if you change the password, to be honest.