StephanJoubert/home_assistant_solarman #107: Corrupted data when restarting HA

arizzi, 2022-10-02:
I noticed that (at least once) when restarting HA many entities went to off the scale values (i.e. corrupted), see attached figure
[image]

would it possible to add some boundary checks and abort the entity update if some variables are out of the expected boundaries? (i.e. a corruption detection)

alienatedsec, 2022-11-03:
Since I used the master version, I have no spikes like yours anymore https://github.com/StephanJoubert/home_assistant_solarman/issues/120#issuecomment-1296209129

[image]

PSXDupe, 2023-01-13:
I am experiencing the same issue more and more.
Is there any recommend solution.