StephanJoubert/home_assistant_solarman #111: Zero values on Total Production leads to wrong energy counting

birdie1, 2022-10-07:
I have a Deye SUN600G3. I am using the inverter_definition deye_string.yml. When the inverter starts in the morning the "Total Production" get a zero value. Because it is from class total_increasing this leads to totally wrong values in energy counting.

It probably should give no data instead of zero values on sensors with class total_increasing.

[image]
[image]

rolandsteinmeyer, 2022-10-12:
see issue #66

birdie1, 2022-10-20:
Unfortunately this does not help, as the inverter reports one 0 value after booting in the morning. The problem is not the unknown value. A possible solution would be, that a type "total_increasing" value should never be allowed to be zero.

alienatedsec, 2022-10-30:
#122 #120 hopefully #125 can fix this

chawawa, 2022-10-30:
> #122 #120 hopefully #125 can fix this

Thank for the update. Will add the code and see how it does. Lets hope this gets merged into the main code.

alienatedsec, 2022-11-01:
> > #122 #120 hopefully #125 can fix this
> 
> 
> 
> Thank for the update. Will add the code and see how it does. Lets hope this gets merged into the main code.
@chawawa  how it went for you? Mine is still the same

Panoramiac, 2022-11-14:
I guess I have the same issue. Each morning, I have a peak in the energy dashboard, which makes no sense. Here is my graph: 
[image]

I am currently removing this peak with the statistics correction tool in the developer tools...

bbjunkie, 2022-11-15:
``` availability: "{{ 10 < states('sensor.solis_total_energy_exported')|float(0) < 20000  }}" ```

This appears to work for me in HACS, limits it to values above 10 and below 20000

Need a proper fix though as its just a kludge

Panoramiac, 2022-11-16:
> availability: "{{ 10 < states('sensor.solis_total_energy_exported')|float(0) < 20000 }}"

Thanks!
Is this a part of a template sensor or where to put that?

bbjunkie, 2022-11-17:
Yes, drop it in the template

bbjunkie, 2022-11-17:
> > availability: "{{ 10 < states('sensor.solis_total_energy_exported')|float(0) < 20000 }}"
> 
> Thanks! Is this a part of a template sensor or where to put that?

Yes, drop it in the template 👇

`- sensor:
    - name: "Filtered Total Energy Exported"
      icon: mdi-counter
      unit_of_measurement: "kWh"
      state: "{{ states('sensor.solis_total_energy_exported')|float(0) }}"

      availability: "{{ 10 < states('sensor.solis_total_energy_exported')|float(0) < 700  }}"
      device_class: energy
      state_class: "total_increasing"`

closed 2022-12-13 by StephanJoubert