StephanJoubert/home_assistant_solarman #114: Sensors stay in "unknwon" status

otello86, 2022-10-11:
Hi,
i try to use this integration with my Sofar HYD-5000ES with wifi logger (S/N: 23XXXXXX). I put on all the parameters and restart HA but all the sensor stay in status “unknown”.
I try different .yaml configuration but the sensors still have unkown status
I don’t have any error message in log and i don't know if the connection to the datalogger works or not
I also try autodiscover mode but without success

The integration is in version 1.4.1
HA works HA OS in version 2022.10.2

Magno06, 2022-11-19:
Same problem. I still don't know how to fix

kkapuscinski, 2022-11-28:
Hi, same here but i have Sofar 8.8 KTLX-G3. for now i see that different port gives "Connection refused" and 8899 stalls for 40 seconds.

vegahansen, 2023-01-04:
I have the same issue. Afore inverter with Solarman logger. Able to log in on the IP adress, hacks reports connected but no values on the sensors in HA. Logger SN 17xxxxxx

s0x00c, 2023-01-07:
Another ME3000SP here. 
Device serial 23xxxx
Telnet connection can be made on port 8899 from the home assistant cli interface.
40 entities are created but I don't seem to be able to gather data.
Willing to test if wanted / needed.

s0x00c, 2023-01-07:
@mcasteels : there is a sofar_wfikit.yaml file included when you clone the repo
I removed the solarman integration completely and then redownloaded it again.
This time, before adding it to the interface, I copied the sofar_wifikit.yaml over the sofar_lsw3.yaml and then proceeded by adding the integration again.
It seems to read some data now from my inverter seeing this in the logbook:
[image]
Then I can click through and see the state of the battery charge during the full last day, meaning it connected and read historic data:
[image]
Right now, I added the counters to the battery dashboard but they don't seem to update yet anyway.
I am seeing this as good news, seems there is a possibility to read the data of the inverter while all other have failed for me.

mcasteels, 2023-01-08:
@s0x00c ,

Thanks for that great info.  I was using the HACS integration and that is different than the one on the site!  I always assumed that these are the same. 
I had first some issues but ended up doing the following and that was a win.  No shortcuts, tried it and ended up to be more work.  So really step by step (again thanks to your info)

1. Remove the integration from HA.
2. Restart HA
3. Remove the integration from HACS (did not do that before and this is also a crucial step, I copied it over the existing before, did not work!).  Verify the custom_components folder that the Solarman folder is deleted.
4. Restart HA
5. Download the GitHub repository (clone) and place it in custom_components
6. Rename the file as you did (you could choose the wifi yaml, but who cares, I did not want to test anymore 😬)
7. Restart HA
8. Add the integration fill in the correct info (it adds 51 entities)

I can see the logbook entries and connection is made and noticed some updates (SN, software version, …).  Verified entity by entity and I do see info.  No more unknowns for the moment.  Still need to verify the data versus the API.  Looks promising.

Before I had several issues trying your solution.  What changed it in the end … Delete all the Solarman integrations (HA and HACS! I also had a disabled one) and restarted after each change and not accumulate to only restart once.

s0x00c, 2023-01-09:
@mcasteels 
Do you have a possibility to add to the energy dashboard?
I see this link: https://github.com/StephanJoubert/home_assistant_solarman/blob/main/energy.md but that definition is not in the sofar_wifikit.yaml.
Would it be possible just to copy those entries over and see if they pick up anything?

s0x00c, 2023-01-09:
Well, I tried that, but copying in the definitions from another yaml config file doesn't work.
We would need the exact modbus "addresses" to read the correct values.

mcasteels, 2023-01-09:
> @mcasteels Do you have a possibility to add to the energy dashboard? I see this link: https://github.com/StephanJoubert/home_assistant_solarman/blob/main/energy.md but that definition is not in the [...]

I was able to add entries to the Energy Dashboard, but really limited and I think that the ones I could add (daily solar panel) will not add anything useful.  I can't add the battery system.

mcasteels, 2023-01-09:
> Well, I tried that, but copying in the definitions from another yaml config file doesn't work. We would need the exact modbus "addresses" to read the correct values.

that is what I tried before and nothing helped.  the only way I could get it working now is thanks to your info.  To be honest I even did not test the sofar_lsw3.yaml.  I just used the sofar_wifikit.yaml but renamed it to the sofar_lsw3.yaml and point to that one when setting up the integration.  Seems that these SOFAR's are not always defined in the same way.  Mine is just used for the battery system and has sensors too look at my solar panel system (that was already installed +10 years ago).  The battery was installed 2 months ago.

s0x00c, 2023-01-09:
> that is what I tried before and nothing helped. the only way I could get it working now is thanks to your info. To be honest I even did not test the sofar_lsw3.yaml. I just used the [...]

Same here, I would like to see the battery integrated in the 'energy dashboard'.
The funny thing is, I can actually see  the battery charging / discharging graphic, but no way to add it to the energy dashboard in separate counters
[image]

mcasteels, 2023-01-09:
Same thing here.

vegahansen, 2023-01-09:
> Well, I tried that, but copying in the definitions from another yaml config file doesn't work. We would need the exact modbus "addresses" to read the correct values.

Yes, I think this is also my issue. Currently I have no information on the adress list for my setup with AFORE inverter. Can't find this anywhere.

s0x00c, 2023-01-13:
@mcasteels 
So, eventually I found the solution:
We needed to split up the charge / discharge counter with a riemann calculation: (in the configuration.yaml file)

```
template:
  - sensor:
    - name: "Battery Charge Power"
      state_class: "measurement"
      device_class: "power"
      unit_of_measurement: "kW"
      icon: mdi:battery-plus
      state: >-
        {% set e = states('sensor.solarman_battery_charge_discharge_power') %}
        {{ e|float(0)|abs if e|float(0) >= 0 else 0 }}
      availability: >-
        {{ states('sensor.solarman_battery_charge_discharge_power')|is_number }}
    - name: "Battery Discharge Power"
      state_class: "measurement"
      device_class: "power"
      unit_of_measurement: "kW"
      icon: mdi:battery-minus
      state: >-
        {% set e = states('sensor.solarman_battery_charge_discharge_power') %}
        {{ e|float(0)|abs if e|float(0) <= 0 else 0 }}
      availability: >-
        {{ states('sensor.solarman_battery_charge_discharge_power')|is_number }}
```
and get that data then in the energy dashboard with adding this to the configuration.yaml file

```
sensor:
  - platform: integration
    source: sensor.battery_charge_power
    name: battery_charge_energy
    round: 2
  - platform: integration
    source: sensor.battery_discharge_power
    name: battery_discharge_energy
    round: 2
```
This gives you everything you need to get the data in the energy dashboard

mcasteels, 2023-01-13:
@s0x00c 
Will have to try it out.  Can I just copy paste this in the configuration.yaml?

s0x00c, 2023-01-13:
Yes, in reverse order than I posted it

mcasteels, 2023-01-15:
@s0x00c 
I'm  noob concerning programming!
Just Copied the data as you mentioned.
First the last part to define the sensor, then the template.  They are both together in one section of the configuration.yaml.  I do have other sensors created Currently no go.  See below screenshot.

[image]

The sensor in the template that you use is exactly the same name as in my config.

[image]

Any idea what I'm doing wrong?

s0x00c, 2023-01-18:
This is what I have in configuration.yaml
[image]
I haven't changed anything else I think.
Same as you, I am NO programmer. I just copy paste and adjust where needed

mcasteels, 2023-01-18:
@s0x00c 
I don’t know why.  Did not check it since I tried and it works 🫣.  Thanks again for the share of info. Only thing I still want to figure out is do isolate some values as W instead of KW.  All other sensors of equipment are published in W.  Looks weird when you put several together in one card.

mcasteels, 2023-01-19:
Info have weird numbers so I’m wondering what you added as input

For Solarpanels I’m using 

[image]

and the battery with your config

[image]

s0x00c, 2023-01-19:
That is where my dashboards differs from yours.
I get my solarpanels info from my SMA device, NOT from solarman, so I won't be able to help you on that front, I am sorry.

mcasteels, 2023-01-19:
No problem, battery settings are similar to yours? Actualy solar looks look it's rather the battery in combination with the rest.

s0x00c, 2023-01-19:
yes, my settings are the same for the battery as yours.

rmupfumira, 2023-01-19:
I have sunsynk hybrid inverter, followed every step in the READM but still entities are unavailable. In the logs I get a connection refused error. IP address of wifi datalogger is correct but am not sure about the port [8899] because it doesn't show as open.
[image]

mayor89, 2023-05-07:
> Hi, same here but i have Sofar 8.8 KTLX-G3. for now i see that different port gives "Connection refused" and 8899 stalls for 40 seconds.

Hello. Have you please managed to solve the issue?